riscvarchive/riscv-gcc #278: [RFC] How to add vendor customized ISA into official toolchain ?

rjiejie, 2021-06-24:
From gcc developer view, there is only one SPN for almost instructions, so

vender must modify the original expand patten if they want to expand customized insn,
the code is ugly now if more than one venders have this issue :(

Is there any reference or guide for this issue ?

jim-wilson, 2021-06-24:
SPN?

You can try looking at other gcc ports for how this was done.  Most won't have the problem as bad as RISC-V, but the mips port is close to the RISC-V and has a lot of different ISA variants.  It is best if you can try to make the support generic, e.g. add a hook for RISC-V target with load-increment support instead of making that Alibaba specific.  Add a hook for RISC-V targets with reg+reg address modes.  Etc.  If the assembly code is different, then that can be handled in the final define_insn, but the rest of the support should be able to be more generic.  The TARGET_SFB_ALU macro is an example of where we added support for a SiFive part and tried to make the implementation generic so others could use it.  For ISA issues, the ISA_HAS_* macros in the MIPS port are probably a better way to do this.

The worst case situation is what happened to ppc/altivec and e500/SPE.  They were originally one gcc port, and then were eventually splits into two parallel gcc ports because they were too difficult to maintain as one port, particularly due to the differences between the Altivec and SPE SIMD extensions.  I think we will have the same situation and need the same solution if we have to support both rvv 0.7.1 and rvv 1.0.

Another option is to take advantage of the fact that if you have two patterns with the same name, then the first one is used and the second one is ignored.  So you could put a (include "alibaba.md") near the top of riscv.md and then just redefine stuff as necessary and the same named patterns in riscv.md won't be used.  That gets us into a duplicate maintenance problem though, were simple changes may require changes to multiple *.md files, so this may not be a good idea.

rjiejie, 2021-06-25:
SPN stands for Standard Pattern Name :)

Thanks for your advice