riscvarchive/riscv-gcc #280: __fp16 is ambiguous error in C++

rjiejie, 2021-06-25:
`#include <math.h>

__fp16 foo (__fp16 a, __fp16 b)
{
        return a + std::exp(b);
}`

compiler options:
=============

riscv64-unknown-linux-gnu-g++ foo.c -march=rv64gc_zfh -mabi=lp64

error:
====

foo.c: In function '__fp16 foo(__fp16, __fp16)':
foo.c:6:23: error: call of overloaded 'exp(__fp16&)' is ambiguous
    6 |  return a + std::exp(b);
      |                       ^
In file included from $INSTALL/sysroot/usr/include/features.h:465,
                 from $INSTALL/riscv64-unknown-linux-gnu/include/c++/10.2.0/riscv64-unknown-linux-gnu/bits/os_defines.h:39,
                 from $INSTALL/riscv64-unknown-linux-gnu/include/c++/10.2.0/riscv64-unknown-linux-gnu/bits/c++config.h:518,
                 from $INSTALL/riscv64-unknown-linux-gnu/include/c++/10.2.0/cmath:41,
                 from $INSTALL/riscv64-unknown-linux-gnu/include/c++/10.2.0/math.h:36,
                 from foo.c:2:
$INSTALL/sysroot/usr/include/bits/mathcalls.h:95:1: note: candidate: 'double exp(double)'
   95 | __MATHCALL_VEC (exp,, (_Mdouble_ __x));
      | ^~~~~~~~~~~~~~
In file included from $INSTALL/riscv64-unknown-linux-gnu/include/c++/10.2.0/math.h:36,
                 from foo.c:2:
$INSTALL/riscv64-unknown-linux-gnu/include/c++/10.2.0/cmath:222:3: note: candidate: 'constexpr float std::exp(float)'
  222 |   exp(float __x)
      |   ^~~
$INSTALL/riscv64-unknown-linux-gnu/include/c++/10.2.0/cmath:226:3: note: candidate: 'constexpr long double std::exp(long double)'
  226 |   exp(long double __x)
      |   ^~~

I think there is no prototype of __fp16 in libmath of glibc,
I could cast '__fp16' to 'float' or 'double' to fix this issue with modifying code, it's not invisible for developers :(

Is there any other method to fix this ?
maybe there is some c++ compiler's option for this ?

jim-wilson, 2021-06-25:
Until the C/C++ standards catch up and define a half float type, anything involving half float is going to be screwed.  We can have only half solutions.  There have been a lot of debates and a lot of disagreements over how this stuff should work.  Anything we pick now will likely have to be changed later.  Also, clang and gcc don't handle half float types exactly the same, which is yet another problem we can't fix until standards say what the right answer is.  Meanwhile, I think trying to add half float support to glibc now may just cause more problems later.  C and newlib are easier targets for this, easier to hack stuff in, and less likely to cause problem if it changes in an incompatible way later.  Embedded folks are used to recompiling everything from scratch anyways.

There is a psABI pr that discusses the large number of problems here.
https://github.com/riscv/riscv-elf-psabi-doc/pull/172

In that thread I talked abourt a short float proposal, but apparently that was killed on the grounds that there is more than one 16-bit FP binary format, and hence short float is ambiguous.  I find that unfortunate and curious.  float and double are ambiguous also, as they map to different float types on different machines.

kito-cheng, 2022-08-01:
We decide use `_Float16` as formal RISC-V half-precision floating point type and included in psABI now https://github.com/riscv-non-isa/riscv-elf-psabi-doc/pull/172