riscvarchive/riscv-gcc #282: Too much 'vsetvli' instuctions

lzwButton, 2021-07-12:
Hello! 
I have tried the latest GCC which supports the latest rvv intrinsic functions API and find that it adds many 'vsetvli' instructions. But actually, all these 'vl' setting instructions are not needed and should be optimized.
Take [rvv_saxpy.c](https://github.com/riscv/rvv-intrinsic-doc/blob/master/rvv_saxpy.c) as example. Here is part of 'saxpy_vec' function I compiled.
```
00000000800004b6 <saxpy_vec>:
    800004b6:	c90d                	beqz	a0,800004e8 <saxpy_vec+0x32>
    800004b8:	013577d7          	vsetvli	a5,a0,e32,m8,tu,mu
    800004bc:	0137f757          	vsetvli	a4,a5,e32,m8,tu,mu
    800004c0:	00279713          	slli	a4,a5,0x2
    800004c4:	0205e407          	vle32.v	v8,(a1)
    800004c8:	95ba                	add	a1,a1,a4
    800004ca:	0137f6d7          	vsetvli	a3,a5,e32,m8,tu,mu
    800004ce:	02066807          	vle32.v	v16,(a2)
    800004d2:	0137f6d7          	vsetvli	a3,a5,e32,m8,tu,mu
    800004d6:	a3055457          	vfmadd.vf	v8,fa0,v16
    800004da:	0137f6d7          	vsetvli	a3,a5,e32,m8,tu,mu
    800004de:	02066427          	vse32.v	v8,(a2)
    800004e2:	8d1d                	sub	a0,a0,a5
    800004e4:	963a                	add	a2,a2,a4
    800004e6:	f969                	bnez	a0,800004b8 <saxpy_vec+0x2>
    800004e8:	8082                	ret
```
There are at least two 'vsetvli' instructions are useless.

jim-wilson, 2021-07-19:
A known problem.  There has been no significant work on the gcc vector port in over a year.  All of the current work is in LLVM which should be used instead of gcc for RISC-V Vector support.

fanghuaqi, 2021-07-20:
Hi @jim-wilson , currently gcc is still important for most of the community, is there any plan to make any improvements on this known issue, or if we want to improve it, could you give us some hints on how to?

Thanks

lzwButton, 2021-07-20:
Thank you for your commit. I wonder if there a plan to support RISC-V Vector. Because currently, GCC is much more stable than LLVM and most customers are familiar with GCC. Thus, they prefer to compile RISC-V vector project with GCC compiler. We are willing to improve it but it is hard to solve this problem. Could you give us some hit on how to?

jim-wilson, 2021-07-20:
I think it needs a few man years of work to be usable.  If you aren't already experienced gcc developers, then maybe a few more man years to learn how to do the work.  It isn't up to date with the evolving RVV ISA spec, it isn't up to date with the evolving RVV intrinsics spec, there are ugly hacks in the vectorizing optimization passes required to make it work, there is no autovectorization support, it is missing basic optimizations like eliminating duplicate vsetvli instructions, etc.  The current status is that it is only useful as a toy for demos.  SiFive and a few other organizations are contributing to the LLVM vector support, but no one is contributing to the gcc vector support.  Alibaba has expressed some interest in contributing recently but it isn't clear how we will handle their patches yet.  The current stuff was mostly done by SiFive, but SiFive is not currently interested in funding this work.

I don't do LLVM work, I only do GCC work.  But this is a very large project to make this work well and be clean enough to upstream.

fanghuaqi, 2021-07-20:
Thanks for your clear explaination, I was not a gcc developer, just a user of it, I previously thought the rvv support in gcc is almost ready to use, now it is still have a long way to go, and of course with hard work done by your team in SiFive which already enable the rvv support is a great base for the community.