riscvarchive/riscv-gcc #298: Hardcoded register numbers macro does not conform to manually specified callee-saved registers

IdanHo, 2021-09-09:
The users of the macro defined in https://github.com/riscv-collab/riscv-gcc/blob/riscv-gcc-11.1.0/gcc/config/riscv/riscv.h#L574 do not support manually specifying callee-saved registers via gcc's `-fcall-saved-reg` flag.

jim-wilson, 2021-09-09:
I don't understand what your problem is.  The CALLEE_SAVED_REG_NUMBER macro and the -fcall-saved-reg flag are different unrelated things.  Is there an actual bug that you found?  If so, what is it?  Please give a testcase demonstrating the problem.

Maybe the problem is that -msave-restore and -fcall-saved-reg don't work together?  If that is the problem, then the solution is to disable -msave-restore when -fcall-saved-reg is used, if that isn't already happening.

IdanHo, 2021-09-09:
> I don't understand what your problem is. The CALLEE_SAVED_REG_NUMBER macro and the -fcall-saved-reg flag are different unrelated things. Is there an actual bug that you found? If so, what is it? [...]

Right, the issue is that the save/restore library functions assume t0-t6 are temporary, and not callee-saved, here's a simple reproducer:
```c
int large_function(int x) {
    int a = x + 5;
    int b = x + 6;
    int c = x + 7;
    int d = x + 8;
    int e = x + 9;
    int f = x + 10;
    return a * 2 + b * 3 + c * 4 + d * 5 + e * 6 + f * 7;
}

int main() {
    asm volatile("li t0, 123          \n\t"
                 "li a0, 0            \n\t"
                 "call large_function \n\t"
                 "ret");
}
```
When compiled with `riscv64-unknown-linux-gnu-gcc -Os -msave-restore -fcall-saved-a2 -fcall-saved-a3 -call-saved-a4 -fcall-saved-a5 -fcall-saved-a6 -fcall-saved-a7 -fcall-saved-t0 -fcall-saved-t1 -fcall-saved-t3 -fcall-saved-t4 -fcall-saved-t5 -fcall-saved-t6 -static test.c` the t0 register is clobbered after returning from the function call (as can be observed with a debugger).
Note that I had to add a very large amount of call-saved registers in order to trigger the issue, as this function is much too small to use up a large amount of the temporaries.
Disabling save-restore lib calls with `-mno-save-restore` does solve my issue, so thanks for that :)

closed 2021-09-12 by IdanHo

jim-wilson, 2021-09-13:
I see you closed this already so maybe you don't need a fix?  t0 and t1 are used implicitly by -msave-restore, so can't be call saved when -msave-restore is used.  t0, t1, t2, and t3 are used implicitly by shared library PLTs, so can't be call saved when -fpic is used.  t0 is used implicitly for large frame sizes, so won't work if any functions have a frame size that doesn't fit in an addi immediate.  I think this is the complete list of implicit use conflicts.  I could add code to check for the -fpic and -msave-restore conflicts and emit an error.  The large frame size one is harder to handle, I could emit a warning for that.

IdanHo, 2021-09-13:
> I see you closed this already so maybe you don't need a fix? t0 and t1 are used implicitly by -msave-restore, so can't be call saved when -msave-restore is used. t0, t1, t2, and t3 are used [...]

I closed this since `-mno-save-restore` is sufficient for my use-case, a warning/error for someone who hasn't seen this issue would likely be helpful.

jim-wilson, 2021-09-13:
Note that -fcall-saved-t0 is still unsafe if you have any functions with a frame size over 4KB or so, as gcc will use t0 as a temporary to compute the frame size before subtracting it from sp.  So you end up with something like this
	addi	sp,sp,-32
	sd	s0,24(sp)
	addi	s0,sp,32
	li	t0,-8192
	addi	t0,t0,192
	add	sp,sp,t0
Since we are using t0 in the prologue, we might not be able to save it before we use it.

IdanHo, 2021-09-13:
> Note that -fcall-saved-t0 is still unsafe if you have any functions with a frame size over 4KB or so, as gcc will use t0 as a temporary to compute the frame size before subtracting it from sp. So [...]

Ah that's good to know, thanks, luckily in my case i only need to callee-save t3-t6.