riscvarchive/riscv-gcc #300: different -O3 behavior about volatile qualifier against clang

unicornx, 2021-09-16:
With clang, it meets my expectation that with `-O3`, the routine `task_delay()` is inlined and the "volatile" qualifier for the parameter count take effect and all read/write to the var would be from/to the memory, but not resides in register.

[image]

But when I tested with gcc, I see the different behavior and the “volatile“ is ignored.

[image]

I  tried the gcc 11.1, also the same result.

It makes me confused, I'm not sure if it is a bug for gcc that not handle c standard well or it's just a valid behavior when we do optimization so different compiler can do it from its self understanding. Anyway, though the looping logic has no problem,  but this optimization difference make me get the delay function cost different time duration.

Thanks in adv.

jim-wilson, 2021-09-16:
When gcc does the function inlining, it replaces the function parameter with the argument value, so the volatile int type effectively disappears.  I don't know if that is a bug or not.  We would need a language lawyer for that.  But I would say in general that you are better off using volatile with locals and globals than with function arguments, since volatile only applies to objects, and that function parameter object can disappear with optimization.  This isn't a RISC-V specific problem.  I see the same result with an x86 gcc.

Anyways, the workaround is trivial.
```
void task_delay (int count)
{
  volatile int count_internal = count;
  count_internal *= 50000;
  while (count_internal--);
}
```
If you want a better answer, file a bug report with FSF GCC.

unicornx, 2021-09-16:
> When gcc does the function inlining, it replaces the function parameter with the argument value, so the volatile int type effectively disappears. I don't know if that is a bug or not. We would need [...]

Thank you, Jim. I have verified your suggestion and it does work.  Let's close this issue first.

closed 2021-09-16 by unicornx