riscvarchive/riscv-gcc #313: Inline asm problem in rvv

sh-zheng, 2021-10-30:
Hello, I use the gcc branch riscv-gcc-10.1-rvv-dev, and the compiler error "impossible constraint in 'asm'" occurs with the following code. I suspect the asm constraint code 'v' raise the error. Do we have supported the code 'v' in this branch?

    vfloat32m1_t v0, v1, v2;

	asm volatile(
        "vsetvli      zero, zero, e32, m1\n\t"
        "vfadd.vv     %0, %1, %2\n\t"
        :"=v"(v0)
        :"v"(v1), "v"(v2)
        :);

kito-cheng, 2021-10-31:
GCC and LLVM both are using `vr` for vector register and `vm` for masking operand.

sh-zheng, 2021-10-31:
> GCC and LLVM both are using `vr` for vector register and `vm` for vector masking operand.

That works. Thank you very much.

kito-cheng, 2021-11-01:
Seems like you got the answered, so I close this issue :)

closed 2021-11-01 by kito-cheng