riscvarchive/riscv-gcc #314: bit field access for hw register

lewis1573, 2021-11-01:
To access HW register, that  can only accessed by 32bit,  i used below code 
```
#define DDR_CTL_BASE 0x20000000
#define DDR_CTL_reg  ((DDR_CTL_Type *) DDR_CTL_BASE) 

typedef struct { 
    union {
        volatile unsigned int reg0; 
        struct {
            volatile unsigned int field0:5;
            volatile unsigned int rsvd_5:11;
            volatile unsigned int field1:8;
            volatile unsigned int rsvd_24:8;
        } reg0_b;       
    };
}DDR_CTL_Type;
int reg_bit_field_access() 
{
    DDR_CTL_reg->reg0_b.field0 = 5;
    DDR_CTL_reg->reg0_b.field1 = 7;
    return 0;
}
```
as I understand,  narrow_volatile_bitfield default  false, so it will access by 32bit, from (https://github.com/riscv-collab/riscv-gcc/issues/248#issuecomment-802949268)
 ` DDR_CTL_reg->reg0_b.field0 = 5;` works as expected, but ` DDR_CTL_reg->reg0_b.field1 = 7;` accessed by 8bit, it used sb 
The compiled assembly code is as follows:
```
reg_bit_field_access():
        addi    sp,sp,-16
        sd      s0,8(sp)
        addi    s0,sp,16
        li      a5,536870912
        lw      a4,0(a5)
        andi    a4,a4,-32
        ori     a4,a4,5
        sw      a4,0(a5)
        li      a5,536870912
        li      a4,7
        sb      a4,2(a5)
        li      a5,0
        mv      a0,a5
        ld      s0,8(sp)
        addi    sp,sp,16
        jr      ra
```
Code and compilation env： https://godbolt.org/z/GKr6866TG
I tried multiple compilers, all gcc compilers have this problem, but clang has no problem
Is there a mistake in my understanding, or is there a bug here?
What can I do to get 32-bit access？
thank you in advance.

jim-wilson, 2021-11-01:
GCC will not change the size of a memory access when volatile is used.  However, you are using volatile with bit-fields.  We don't hvae 7-bit read/write instructions, so we must use a larger size, which is arbtrarily set to 32-bits.  We do have 8-bit read/write instructions so we use them.

You have contradictory requirements here.  For the 7-bit bit-field you are happy that the size of the access was changed.  For the 8-bit bit-field, you are  unhappy that the size wasn't changed.  I don't think that clang guarantees that you will always get the 32-bit access you want for both objects.  I think this is just an accident of optimization.  If you really want 32-bit accesses, then you must use a 32-bit type, and use masks and shifting to insert the bit-field values.

Internally in gcc, narrow_volatile_bitfield will control the size of an access for a bit-field when we can't directly access it.  This is false by default, and not changed by the RISC-V backend, so this is why the 7-bit access gets widened to 32-bits instead of just 8-bits.  However, for the 8-bit bit-field, because it is the same size and alignment as a char, the gcc optimizer treats it exactly the same as a char field.  Hence, the narrow_volatile_bitfield check does not apply, and it does not get widended to 32-bits.  This isn't a RISC-V issue.  This happens for all targets.

lewis1573, 2021-11-03:
thank you so much!
so sand we cannot use the beautiful struct to access register

closed 2021-11-08 by lewis1573