riscvarchive/riscv-gcc #320: Is there a stable gcc10.2 branch which support RVV 1.0

fanghuaqi, 2021-12-14:
Hi there, 

     Is there a gcc10.2 branch maintained provide good support for rvv 1.0(with intrinsic API support). I found https://github.com/riscv-collab/riscv-gcc/tree/riscv-gcc-10.1-rvv-dev this branch is actively developed but it is gcc 10.1 based and I didn't know the status of this branch. Could anybody kindly point out it for me?

Thanks

jim-wilson, 2021-12-14:
There is no actively maintained gcc rvv support, and no ongoing gcc rvv development.  Current work is all in LLVM, and LLVM is recommended if you want rvv support.  There is a guy at rivai.ai who claims to be working on gcc rvv 1.0 support for mainline FSF gcc, so maybe in a few months we will have something usable.  Rivai is in Shenzhen.

fanghuaqi, 2021-12-14:
Thank you @jim-wilson , I will take a look at the upstream gcc, so the currently developed branch `riscv-gcc-10.1-rvv-dev` is not yet tested and proven?

jim-wilson, 2021-12-14:
SiFive abandoned the gcc rvv work and is doing only llvm rvv work now.  The gcc rvv branch is badly out of date.  It should be mostly working, but is not compatible with rvv 1.0 or the current rvv intrinsics spec.

rjiejie, 2021-12-15:
> Hi there,
>
> Is there a gcc10.2 branch maintained provide good support for rvv 1.0(with intrinsic API support). I found https://github.com/riscv-collab/riscv-gcc/tree/riscv-gcc-10.1-rvv-dev this [...]

We have also supported/maintained the RVV v1.0 feature, you could download prebuilt gcc toolchain from Alibaba website[1].

[1] : https://occ.t-head.cn/community/download?id=3988303250540531712

lhtin, 2021-12-17:
> We have also supported/maintained the RVV v1.0 feature, you could download prebuilt gcc toolchain from Alibaba website[1].
>
> [1] : https://occ.t-head.cn/community/download?id=3988303250540531712

Do you have any plan to open source the RVV v1.0 feature?

TommyMurphyTM1234, 2022-10-11:
Just to update this - Vector extension support is now being carried out on the `rvv-next` branch.
Example build:
```
git clone https://github.com/riscv/riscv-gnu-toolchain rv64-linux-rvv-next -b rvv-next
cd rv64-linux-rvv-next
./configure --prefix=`pwd`/installed-tools --with-arch=rv64gcv
date > build.log; make linux 2>&1 | tee -a build.log; date >> build.log
```

fanghuaqi, 2022-10-14:
Thanks @TommyMurphyTM1234 for the update

closed 2022-10-14 by fanghuaqi

troibe, 2023-11-30:
In case I (or someone else) run into this issue again.
The master branch of this repo has now support for the vector extension and the comment about `rvv-next` is irrelevant.

lldb123, 2024-01-04:
> Just leave few more comment for anyone who read this issue in future.
> 
> DO NOT USE `rvv-next` branch, it's _deprecated_, RVV intrinsic is supported since GCC 13 and auto vectorziation will supported in GCC 14.

Hello, I am currently using the main branch of this repo to compile riscv-gcc and it already can auto vectorize. 

When you say "RVV intrinsic is supported since GCC 13 and auto vectorziation will supported in GCC 14", does it mean GCC will supporting Vector Length Agnostic (VLA) vectorization?