riscvarchive/riscv-gcc #324: GCC emits the wrong symbols for the "Routines for integer arithmetic"

echelonxray, 2022-01-26:
When GCC needs to invoke one if it's Routines for integer arithmetic, it emits the wrong symbol for the types concerned.  The parameters are setup in the registers correctly per the ABI for the types concerned.  However, since the actual function it called is wrong, it can end up getting garbage.  I found this to also be the case when building for rv64.  However I only tested my demo program which uses unsigned types.  I don't know that any problems will occur there because it always calls either __udivdi3 or __umoddi3 and those will match the XLEN of 64 bits.  For unsigned types, it might work.  Although, I'm tempted to say this is not the intended behavior.

[gcc_bug.c.zip](https://github.com/riscv-collab/riscv-gcc/files/7944648/gcc_bug.c.zip)

In the case of my example here, it called __udivdi3 and __umoddi3 instead of __udivti3 and __umodti3.  Per the ABI, __udivdi3 and __umoddi3 interpreted the low 32 bits of the first parameter to be argument 1, and the high 32 bits to be argument 2.  This causes a division or modulus by 0 attempt.  

See the attached demo program.  Build it with (Replace the riscv*-gcc with whatever the triple is on your system): 
`riscv*-gcc -Wall -Wextra -std=c99 -march=rv32ia -mabi=ilp32 -mcmodel=medany -O0 -mrelax  -fno-stack-check -fno-stack-protector -fomit-frame-pointer -ftls-model=local-exec -fno-pic -ffreestanding -nostdlib -nostartfiles -fno-zero-initialized-in-bss -e main ./gcc_bug.c -o ./gcc_bug.out`

The output: 
```
$ qemu-riscv32 ./gcc_bug.out 
GCC Bug Demo Running.
Dividen: 7
Divisor: 2

Divide unsigned ints
Hello from __udivsi3
Modulus unsigned ints
Hello from __umodsi3
c1: 3, d1: 1

Divide unsigned longs
Hello from __udivsi3
Modulus unsigned longs
Hello from __umodsi3
c2: 3, d2: 1

Divide unsigned long longs
Hello from __udivdi3
	b is 0.  Returning with 0.
Modulus unsigned long longs
Hello from __umoddi3
	b is 0.  Returning with 0.
c3: 0, d3: 0

[Work Around] Divide unsigned long longs
Hello from __udivti3
[Work Around] Modulus unsigned long longs
Hello from __umodti3
c4: 3, d4: 1

```

kito-cheng, 2022-01-27:
`di` means 64 bits integer, but you declare `__umoddi3` and `__udivdi3` with `unsigned long`, `unsigned long` is 64 bit on RV64, but it's only 32 bit on RV32, so that's why you got unexpected result.

```
unsigned long __umoddi3(unsigned long a, unsigned long b);
unsigned long __udivdi3(unsigned long a, unsigned long b);
```

And another issue is `ti` means 128 bit integer, and `long long` is just 64 bit on both RV32 and RV64, so it should be `__int128` rather than `long long`, but `__int128` is unsupported on RV32.

echelonxray, 2022-01-27:
That would seem to conflict with this: https://gcc.gnu.org/onlinedocs/gccint/Integer-library-routines.html
And the datatype sizes specified on page 2 here: https://riscv.org/wp-content/uploads/2015/01/riscv-calling.pdf

If "di" means 64-bit, wouldn't that make the gnu manual page wrong on the case of rv32?  It lists it as dealing with type "long".

jim-wilson, 2022-02-01:
Yes, the gcc integer library routine docs are broken.

The situation is a little more complicated than what Kito mentioned, as __udivdi3 can have different type signatures in different gcc ports.  This could be a 16-bit divide for an 8-bit target, a 32-bit divide for a 16-bit target, a 64-bit divide for a 32-bit target, a 64-bit divide for a 64-bit target, or a 128-bit divide for a 64-bit target.  Some of this is historical error, and some is convenience.

Since the RISC-V port supports both 32-bit and 64-bit code from the same compiler binary, we force the default 32-bit choice which is that udivdi3 is 64-bits, and use that choice for both 32-bit and 64-bit code.

The only safe situation here is to never define or call the gcc internal library functions, because that isn't portable.  And if you can't avoid that, then the code needs to be conditional on the target you are compiling for.

If we want a portable ABI, then we have to define our own instead of using the gcc one.  ARM did this for instance.  And there was a proposal on the old EABI tech mailing list for a RISC-V equivalent.

echelonxray, 2022-02-02:
Is there a place where this behavior of the port is documented?  Perhaps in the ABI?

jim-wilson, 2022-02-02:
This is considered gcc internals.  No one other than gcc (and maybe llvm?) should be calling these functions.  Though once a port makes a choice, it can't change it without breaking backward compatibility, so this choice should never change.

If we want documentation, we should have a RISC-V ABI, not a gcc internal ABI.  The EABI proposal for instance defines
```
regstruct32_t __rvabi_idivmod( int32_t n, int32_t d)
```
where the struct contains both the div and mod results.

There are also functions
```
regstruct32_t __rvabi_uidivmod( uint32_t n, uint32_t d)
regstruct64_t __rvabi_ldivmod(int64_t n, int64_t d)
regstruct64_t __rvabi_uldivmod(uint64_t n, uint64_t d)
```

echelonxray, 2022-02-02:
I see...these are proposals for a possible RV ABI standard.  I think that right now it is a problem as freestanding projects, targeting systems without M-extension support will need to implement these.  Right now, the Linux kernel does not support building without M-support.  However, I'd eventually like to patch that.  Currently, I'm emulating in firmware.  

I have also had problems building GCC to target rv32ia.  Using the build scripts provided by the riscv-gnu-toolchain, it seems to work correctly.  But trying to manually build using Vanilla Upstream GCC releases with variants of the LFS book's instructions result libgcc not being built with the specified MARCH.  Instead, it seems to fall back to some default and is built with compressed instructions.  Programs built/linked against that tool-chain won't work on rv32ia.

echelonxray, 2022-02-02:
Gentoo's crossdev has the same problem.  It produces a libgcc with compressed instructions.

jim-wilson, 2022-02-02:
I'm not familiar with the LFS book or gentoo.  But if you are trying to build with --enable-multilib, then we only build a predefined set of multilibs, and rv32ia is not one of them, but rv32iac is.  There are many thousands, maybe millions, of possible combinations of RISC-V ISA extensions, and we can't build them all by default.  The solution is to build the one you need using --with-arch/--with-abi.  Or add the one you need to the set of standard multilibs.    There is a script gcc/config/riscv/multilib-generator you can run to generate the list of multilibs you want.  See the t-elf-multilib file for the default values used.  There is a configure option --with-multilib-generator= that you can use to specify the list at configure time instead of modifying gcc sources.

echelonxray, 2022-02-08:
Thank you for your help.  I was able to make it further in creating a cross-compiler with the --with-arch/--with-abi configure options.  It's off topic from the subject of this thread so I will follow up elsewhere if I have further issues.  

As it stands, I have begun patching my firmware and freestanding projects to use xint32_t and xint64_t for __x???si and __x???di respectively on risc-v.  I hope some standard can be reached and documented in reguard to this.  Otherwise, freestanding projects will be forced to use gcc "internal" symbol names in a manner which is likely to be wrong for the arch (given that the docs are wrong).  However, my issue/question has been answered and I have a working solution.  I will close this issue.

closed 2022-02-08 by echelonxray