riscvarchive/riscv-gcc #325: Failure in make install newlib [commit=149e217033f01410a9783c5cb2d020cf8334ae4c] Minimal support of Zba,Zbb,Zbc,Zbs bitmanip instructions in gcc 12

richunorman-1, 2022-01-28:
On 06 January 2022 I git clone riscv-gcc repository and checkout the following riscv-gcc which claim to provide minimal support of Zba,Zbb,Zbc,Zbs bitmanip instructions in gcc 12. Also the given branch binutils by following the steps:
**Steps**
git clone https://sourceware.org/git/binutils-gdb.git
 cd binutils-gdb &&  git checkout 9455c91957590ca6d4520cfe0955f9f9f1349f82

git clone https://gcc.gnu.org/git/gcc.git
cd gcc && git checkout 149e217033f01410a9783c5cb2d020cf8334ae4c
[error-on-newlib- gcc12.txt](https://github.com/riscv-collab/riscv-gcc/files/7956751/error-on-newlib-.gcc12.txt)
[image]

I could configure and install both binutils and gcc,then I git clone and checkout riscv-newlib-3.2.0 branch.
It shows the following error on make install (newlib)

riscv32-unknown-elf-ar rc ../libc.a *.o
riscv32-unknown-elf-ar: ../argz/lib.a: No such file or directory
riscv32-unknown-elf-ar: ../stdlib/lib.a: No such file or directory
riscv32-unknown-elf-ar: ../ctype/lib.a: No such file or directory
riscv32-unknown-elf-ar: ../search/lib.a: No such file or directory
riscv32-unknown-elf-ar: ../stdio/lib.a: No such file or directory
riscv32-unknown-elf-ar: ../string/lib.a: No such file or directory
riscv32-unknown-elf-ar: ../signal/lib.a: No such file or directory
riscv32-unknown-elf-ar: ../time/lib.a: No such file or directory
riscv32-unknown-elf-ar: ../locale/lib.a: No such file or directory
riscv32-unknown-elf-ar: ../reent/lib.a: No such file or directory
riscv32-unknown-elf-ar: ../errno/lib.a: No such file or directory
riscv32-unknown-elf-ar: ../misc/lib.a: No such file or directory
riscv32-unknown-elf-ar: ../ssp/lib.a: No such file or directory
riscv32-unknown-elf-ar: ../syscalls/lib.a: No such file or directory
riscv32-unknown-elf-ar: ../machine/lib.a: No such file or directory
riscv32-unknown-elf-ar: *.o: No such file or directory
Makefile:1031: recipe for target 'libc.a' failed
make[4]: *** [libc.a] Error 1
make[4]: Leaving directory '/home/richu/Documents/riscv32_gcc12/riscv-tools/riscv-newlib/riscv-newlib-build/riscv32-unknown-elf/newlib/libc'
Makefile:683: recipe for target 'install-recursive' failed
make[3]: *** [install-recursive] Error 1
make[3]: Leaving directory '/home/richu/Documents/riscv32_gcc12/riscv-tools/riscv-newlib/riscv-newlib-build/riscv32-unknown-elf/newlib/libc'
Makefile:641: recipe for target 'install-recursive' failed
make[2]: *** [install-recursive] Error 1
make[2]: Leaving directory '/home/richu/Documents/riscv32_gcc12/riscv-tools/riscv-newlib/riscv-newlib-build/riscv32-unknown-elf/newlib'
Makefile:8521: recipe for target 'install-target-newlib' failed
make[1]: *** [install-target-newlib] Error 2
make[1]: Leaving directory '/home/richu/Documents/riscv32_gcc12/riscv-tools/riscv-newlib/riscv-newlib-build'
Makefile:2275: recipe for target 'install' failed
make: *** [install] Error 2
[error-on-newlib- gcc12.txt](https://github.com/riscv-collab/riscv-gcc/files/7956736/error-on-newlib-.gcc12.txt)

pz9115, 2022-01-28:
I think it is an error due to configure no set correctly, you can just use the [riscv-gnu-toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain) repo, checkout gcc, binutils, newlib into the commit you mentioned, set configure like [README.md](https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/README.md) suggest to generate a makefile then build the whole program. 

Here is a simple script that build with the lastest GCC with upstream also contain Bitmanip extension us rv64:

https://github.com/plctlab/riscv-ci/blob/main/gnu-toolchain-upstream-rv64.sh

Hope this will help you.

richunorman-1, 2022-01-28:
> I think it is an error due to configure no set correctly, you can just use the [riscv-gnu-toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain) repo, checkout gcc, binutils, newlib into the commit you mentioned, set configure like [README.md](https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/README.md) suggested to generate a makefile then build the whole program.
> 
> And here is a simple script that can build with the lastest GCC with upstream, it also contain Bitmanip extension use rv64:
> 
> https://github.com/plctlab/riscv-ci/blob/main/gnu-toolchain-upstream-rv64.sh
> 
> Hope this will help you.

Thanking you for the Response.
I have installed the toolchain for riscv64. I wish to check with that particular branch of gcc 12 which provide minimal support for bitmanip instructions . It would be helpful if you could support with that.

richunorman-1, 2022-01-28:
> I think it is an error due to configure no set correctly, you can just use the [riscv-gnu-toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain) repo, checkout gcc, binutils, newlib into [...]

I will try with that.

pz9115, 2022-01-28:
> > I think it is an error due to configure no set correctly, you can just use the [riscv-gnu-toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain) repo, checkout gcc, binutils, newlib into the commit you mentioned, set configure like [README.md](https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/README.md) suggested to generate a makefile then build the whole program.
> > And here is a simple script that can build with the lastest GCC with upstream, it also contain Bitmanip extension use rv64:
> > https://github.com/plctlab/riscv-ci/blob/main/gnu-toolchain-upstream-rv64.sh
> > Hope this will help you.
> 
> Thanking you.I will try with that. I have installed the toolchain for riscv64. I wish to check with that particular branch of gcc 12 which provide minimal support for bitmanip instructions . It would be helpful if you could support with that.

You can just modify the script, and reset into the minimal support commit, then it can use.

[script.txt](https://github.com/riscv-collab/riscv-gcc/files/7957139/script.txt)

richunorman-1, 2022-01-30:
> You can just modify the script, and reset into the minimal support commit, then it can use.
>
> [script.txt](https://github.com/riscv-collab/riscv-gcc/files/7957139/script.txt)

Hi,
I have tried the script. Still the error occured while switching to that particular gcc branch
** git reset --hard  149e217033f01410a9783c5cb2d020cf8334ae4c
fatal: not a git repository (or any of the parent directories): .git **

pz9115, 2022-01-30:
> > > > I think it is an error due to configure no set correctly, you can just use the [riscv-gnu-toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain) repo, checkout gcc, binutils, newlib into the commit you mentioned, set configure like [README.md](https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/README.md) suggested to generate a makefile then build the whole program.
> > > > And here is a simple script that can build with the lastest GCC with upstream, it also contain Bitmanip extension use rv64:
> > > > https://github.com/plctlab/riscv-ci/blob/main/gnu-toolchain-upstream-rv64.sh
> > > > Hope this will help you.
> > > 
> > > 
> > > Thanking you.I will try with that. I have installed the toolchain for riscv64. I wish to check with that particular branch of gcc 12 which provide minimal support for bitmanip instructions . It would be helpful if you could support with that.
> > 
> > 
> > You can just modify the script, and reset into the minimal support commit, then it can use.
> > [script.txt](https://github.com/riscv-collab/riscv-gcc/files/7957139/script.txt)
> 
> Hi, I have tried the script. Still the error occured while switching to that particular gcc branch ** git reset --hard [149e217](https://github.com/riscv-collab/riscv-gcc/commit/149e217033f01410a9783c5cb2d020cf8334ae4c) fatal: not a git repository (or any of the parent directories): .git **

I think you are not work in right path, please check it:

[image]

richunorman-1, 2022-01-31:
> I think you are not work in right path, please check it:
>
> [image]

I could switch to that gcc branch thank you

closed 2022-02-04 by richunorman-1