riscvarchive/riscv-gcc #326: Warning on compiling the c file using gcc 12

richunorman-1, 2022-02-07:
I have installed gcc 12 latest snapshot and supporting newlib 4.2,gdb version 11.2 and binutils 2.38 branch.

On compiling I got the following warning.How could i minimize this warning?
**riscv64-unknown-elf-gcc main.c -o main.elf**
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/crtbegin.o: mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/crtbegin.o: mis-matched ISA version 2.0 for 'a' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /tmp/cchUJd6U.o: mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /tmp/cchUJd6U.o: mis-matched ISA version 2.0 for 'a' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-exit.o): mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-exit.o): mis-matched ISA version 2.0 for 'a' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-getchar.o): mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-getchar.o): mis-matched ISA version 2.0 for 'a' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-getc.o): mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-getc.o): mis-matched ISA version 2.0 for 'a' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o): mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-impure.o): mis-matched ISA version 2.0 for 'a' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-init.o): mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-init.o): mis-matched ISA version 2.0 for 'a' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-printf.o): mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-printf.o): mis-matched ISA version 2.0 for 'a' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-rget.o): mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-rget.o): mis-matched ISA version 2.0 for 'a' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-vfprintf.o): mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-vfprintf.o): mis-matched ISA version 2.0 for 'a' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-wsetup.o): mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-wsetup.o): mis-matched ISA version 2.0 for 'a' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-__call_atexit.o): mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-__call_atexit.o): mis-matched ISA version 2.0 for 'a' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fflush.o): mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fflush.o): mis-matched ISA version 2.0 for 'a' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-findfp.o): mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-findfp.o): mis-matched ISA version 2.0 for 'a' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fini.o): mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fini.o): mis-matched ISA version 2.0 for 'a' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-freer.o): mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-freer.o): mis-matched ISA version 2.0 for 'a' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-frexpl.o): mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
[... cut ...]
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/libgcc.a(extenddftf2.o): mis-matched ISA version 2.0 for 'a' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/crtend.o: mis-matched ISA version 2.0 for 'i' extension, the output version is 2.1
/lib/gcc/riscv64-unknown-elf/12.0.0/../../../../riscv64-unknown-elf/bin/ld: warning: /lib/gcc/riscv64-unknown-elf/12.0.0/crtend.o: mis-matched ISA version 2.0 for 'a' extension, the output version is 2.1

kito-cheng, 2022-02-07:
Did you rebuild everything with GCC 12 and binutils 2.38? that seems like newlib is built with ISA spec 2.2?

More detail for why those warning message appear:
https://groups.google.com/a/groups.riscv.org/g/sw-dev/c/aE1ZeHHCYf4

richunorman-1, 2022-02-07:
Yes .
Thanking you. 
Right now I have configured the gcc with ' --with-arch=rv64imac_zba_zbb_zbc_zbs_zicsr_zifencei '. I will try the solution and get back.

closed 2022-02-19 by richunorman-1