riscvarchive/riscv-gcc #338: What is the status of riscv-gcc-rvv-next branch

fanghuaqi, 2022-04-29:
Hi @zhongjuzhe , thanks for the great work on rvv support, could you kindly make some note about the status of riscv-gcc-rvv-next branch you are maintaining?

- [ ] Is rvv intrinsic API supported? As I can observe, the rvv intrinsic API is now on discussion, so I want to confirm which commit of https://github.com/riscv-non-isa/rvv-intrinsic-doc/ you are supporting now?
- [ ] Is rvv auto vector supported now? What features of auto vector are supported? 

Thanks

zhongjuzhe, 2022-04-29:
> Hi @zhongjuzhe , thanks for the great work on rvv support, could you kindly make some note about the status of riscv-gcc-rvv-next branch you are maintaining?
>
> * [ ]  Is rvv intrinsic API supported? [...]
```
I am actively maintaining this branch. All intrinsics including C and C++ in this document:https://github.com/riscv-non-isa/rvv-intrinsic-doc/tree/new-policy-proposal
are supported. You can check the discussion here:https://github.com/riscv-non-isa/rvv-intrinsic-doc/pull/137.
This proposal is base on `master` branch document(only have vop and vop_m intrinsics) and add more intrinsics like vop_ta, vop_tu, vop_tamu, vop_tama,vop_tuma,vop_tumu,vop_tam,vop_tum,vop_ma,vop_mu. 
```
> * [ ]  Is rvv auto vector supported now? What features of auto vector are supported?
```
The answer is yes. you can use `-mrvv` compile option to enable VLA auto-vectorization, the binary can run on any vector-length RVV machine.
Also you can add `mriscv-vector-bits=N` to enable VLS auto-vectorization. But this compile option has a side effect:
the binary which is generated by using "-mrvv -mriscv-vector-bits=128" compile option can only run on vector length=128
RVV machine. But this compile option can achieve the best performance.

I am going to implement `-mriscv-vector-bits-min=N` compile option like `--riscv-v-vector-bits-min=N` in LLVM. And I will open source them in the future.
```
> 
> Thanks

fanghuaqi, 2022-04-29:
Thank you for the details sharing. It solved my question.

closed 2022-04-29 by fanghuaqi

kito-cheng, 2022-04-29:
> I am going to implement `-mriscv-vector-bits-min=N` compile option like `--riscv-v-vector-bits-min=N` in LLVM. 
And I will open source them in the future.

Actually that option will removed from LLVM in future since we already have information from `zvl*b`.

zhongjuzhe, 2022-04-29:
> > I am going to implement `-mriscv-vector-bits-min=N` compile option like `--riscv-v-vector-bits-min=N` in LLVM.
> > And I will open source them in the future.
> 
> Actually that option will removed from LLVM in future since we already have information from `zvl*b`.

Oh, I see. I am going to implement this specified by `zvl*b`.