riscvarchive/riscv-gcc #341: compiling problems at branch "riscv-gcc-rvv-next"

PPParticle, 2022-05-13:
Hello!
Since I noticed the PR #329 which has been accepted on May 1st, I tried using it. However, there is always something wrong happening at compiling.

```
$ git submodule status
 553032db17440f8de011390e5a1cfddd13751b0b qemu (v5.2.0)
+20756b0fbe065a84710aa38f2457563b57546440 riscv-binutils (heads/riscv-binutils-2.38)
 4ea498a8e1fafeb568530d84db1880066478c86b riscv-dejagnu (heads/riscv-dejagnu-1.6)
+8b4f700505676160adbd8ef02ffa5b985e153407 riscv-gcc (heads/riscv-gcc-rvv-next)
 5da071ef0965b8054310d8dde9975037b0467311 riscv-gdb (remotes/origin/fsf-gdb-10.1-with-sim)
 9826b03b747b841f5fc6de2054bf1ef3f5c4bdf3 riscv-glibc (glibc-2.33)
 415fdd4279b85eeec9d54775ce13c5c412451e08 riscv-newlib (newlib-4.1.0)

$ ./configure --prefix="$RISCV" --with-arch=rv64gcv --with-abi=lp64d  
$  sudo make 
$  sudo make | grep error

Assembler messages:
Error: rv64imafdcv_zvamo_zvlsseg: unknown prefixed ISA extension `zvamo'
/tmp/ccppjnGN.s:3: Error: rv64i2p0_m2p0_a2p0_f2p0_d2p0_c2p0_v1p0_zvamo1p0_zvlsseg1p0: unknown prefixed ISA extension `zvamo'
make[6]: *** [Makefile:393: lib_a-dummy.o] Error 1
Assembler messages:
Error: rv64imafdcv_zvamo_zvlsseg: unknown prefixed ISA extension `zvamo'
/tmp/ccdMl3hc.s:3: Error: rv64i2p0_m2p0_a2p0_f2p0_d2p0_c2p0_v1p0_zvamo1p0_zvlsseg1p0: unknown prefixed ISA extension `zvamo'
make[6]: *** [Makefile:632: lib_a-__adjust.o] Error 1
Assembler messages:
Error: rv64imafdcv_zvamo_zvlsseg: unknown prefixed ISA extension `zvamo'
/tmp/ccByduAI.s:3: Error: rv64i2p0_m2p0_a2p0_f2p0_d2p0_c2p0_v1p0_zvamo1p0_zvlsseg1p0: unknown prefixed ISA extension `zvamo'
make[6]: *** [Makefile:537: lib_a-ctype_.o] Error 1
Assembler messages:
Error: rv64imafdcv_zvamo_zvlsseg: unknown prefixed ISA extension `zvamo'
/tmp/ccJ2asb9.s:3: Error: rv64i2p0_m2p0_a2p0_f2p0_d2p0_c2p0_v1p0_zvamo1p0_zvlsseg1p0: unknown prefixed ISA extension `zvamo'
make[6]: *** [Makefile:404: lib_a-bsearch.o] Error 1
Assembler messages:
Error: rv64imafdcv_zvamo_zvlsseg: unknown prefixed ISA extension `zvamo'
/tmp/cc0q93zC.s:3: Error: rv64i2p0_m2p0_a2p0_f2p0_d2p0_c2p0_v1p0_zvamo1p0_zvlsseg1p0: unknown prefixed ISA extension `zvamo'
make[6]: *** [Makefile:820: lib_a-fiprintf.o] Error 1
Assembler messages:
Error: rv64imafdcv_zvamo_zvlsseg: unknown prefixed ISA extension `zvamo'
/tmp/ccm6tje3.s:3: Error: rv64i2p0_m2p0_a2p0_f2p0_d2p0_c2p0_v1p0_zvamo1p0_zvlsseg1p0: unknown prefixed ISA extension `zvamo'
make[6]: *** [Makefile:581: lib_a-bcopy.o] Error 1
Assembler messages:
Error: rv64imafdcv_zvamo_zvlsseg: unknown prefixed ISA extension `zvamo'
/tmp/ccYHIogv.s:3: Error: rv64i2p0_m2p0_a2p0_f2p0_d2p0_c2p0_v1p0_zvamo1p0_zvlsseg1p0: unknown prefixed ISA extension `zvamo'
make[6]: *** [Makefile:346: lib_a-psignal.o] Error 1
Assembler messages:
Error: rv64imafdcv_zvamo_zvlsseg: unknown prefixed ISA extension `zvamo'
/tmp/cc4QfFDY.s:3: Error: rv64i2p0_m2p0_a2p0_f2p0_d2p0_c2p0_v1p0_zvamo1p0_zvlsseg1p0: unknown prefixed ISA extension `zvamo'
make[6]: *** [Makefile:395: lib_a-asctime.o] Error 1
```

I don't know what is going on.  If there is something wrong on my installation procedures, please let me know. Thanks so much!!

kito-cheng, 2022-05-13:
> +8b4f700505676160adbd8ef02ffa5b985e153407 riscv-gcc (heads/riscv-gcc-rvv-next)

The hash seems weird? that `riscv-gcc-10.1-rvv-dev` rather than `riscv-gcc-rvv-next` branch?

pavitrar20, 2022-11-08:
I encountered the following error 
Making all in argz
make[6]: Entering directory '/home/ubuntu/Spike/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/libc/argz'
riscv64-unknown-elf-gcc -B/home/ubuntu/Spike/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/ -isystem /home/ubuntu/Spike/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/targ-include -isystem /home/ubuntu/Spike/riscv-gnu-toolchain/newlib/newlib/libc/include -B/home/ubuntu/Spike/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/libgloss/riscv64 -L/home/ubuntu/Spike/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/libgloss/libnosys -L/home/ubuntu/Spike/riscv-gnu-toolchain/newlib/libgloss/riscv64    -DPACKAGE_NAME=\"newlib\" -DPACKAGE_TARNAME=\"newlib\" -DPACKAGE_VERSION=\"4.1.0\" -DPACKAGE_STRING=\"newlib\ 4.1.0\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -I. -I/home/ubuntu/Spike/riscv-gnu-toolchain/newlib/newlib/libc/argz -fno-builtin -DHAVE_NANOSLEEP      -O2 -D_POSIX_MODE -ffunction-sections -fdata-sections   -mcmodel=medlow -c -o lib_a-dummy.o `test -f 'dummy.c' || echo '/home/ubuntu/Spike/riscv-gnu-toolchain/newlib/newlib/libc/argz/'`dummy.c
riscv64-unknown-elf-gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory
compilation terminated.
make[6]: *** [Makefile:393: lib_a-dummy.o] Error 1

Compilations steps
 ./configure --prefix="$RISCV" --with-arch=rv64gcv --with-abi=lp64d  
$  sudo make 

riscv-gnu-toolchain$ git submodule status
 9826b03b747b841f5fc6de2054bf1ef3f5c4bdf3 glibc (glibc-2.33)
-85e0e3519655220688e757b9d5bfd314923548bd musl
 415fdd4279b85eeec9d54775ce13c5c412451e08 newlib (newlib-4.1.0)
-553032db17440f8de011390e5a1cfddd13751b0b qemu
 b51c2fec1da205ea3e7354cbb3e253018d64873c riscv-binutils (binutils-2_39)
-4ea498a8e1fafeb568530d84db1880066478c86b riscv-dejagnu
+eb96155f49deefcde0b054f6c02aa67c607ac482 riscv-gcc (remotes/origin/riscv-gcc-rvv-next)
 5da071ef0965b8054310d8dde9975037b0467311 riscv-gdb (remotes/origin/fsf-gdb-10.1-with-sim)

I need to test RVV with Spike and PK