riscvarchive/riscv-gcc #344: `-march=rv64gc` and `-march=rv64imafdc` may behave differently with multilib

fangzh-umich, 2022-06-27:
On my system, trying to compile a C++ source file that includes `<string>` (and iirc some other headers) with `-march=rv64gc` results in a compile error:
```
root@8059ca370f34:/programs# cat hello.cpp
#include <string>

int main()
{
  return 0;
}
root@8059ca370f34:/programs# riscv64-unknown-elf-g++ -march=rv64gc hello.cpp
In file included from hello.cpp:1:
/opt/riscv/riscv64-unknown-elf/include/c++/11.1.0/string:38:10: fatal error: bits/c++config.h: No such file or directory
   38 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
root@8059ca370f34:/programs# riscv64-unknown-elf-g++ -march=rv64imafdc hello.cpp  # Compiles successfully
root@8059ca370f34:/programs#
```

Adding the `-v` option to the command line shows that the problem has something to do with multilib: In the first case, the compiler tries to use `/opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/include/c++/11.1.0/riscv64-unknown-elf/rv64imafdc/lp64d` as an include directory, which does not exist. Meanwhile, in the second case, the compiler correctly uses ` /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/include/c++/11.1.0/riscv64-unknown-elf`.
```
root@8059ca370f34:/programs# riscv64-unknown-elf-g++ -march=rv64gc hello.cpp -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-g++
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv64-unknown-elf/11.1.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion= --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././riscv-gcc --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket --with-isa-spec=2.2 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medany'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 11.1.0 () 
COLLECT_GCC_OPTIONS='-march=rv64gc' '-v' '-mtune=rocket' '-mabi=lp64d' '-march=rv64imafdc' '-dumpdir' 'a-'
 /opt/riscv/libexec/gcc/riscv64-unknown-elf/11.1.0/cc1plus -quiet -v -imultilib rv64imafdc/lp64d hello.cpp -quiet -dumpdir a- -dumpbase hello.cpp -dumpbase-ext .cpp -march=rv64gc -mtune=rocket -mabi=lp64d -march=rv64imafdc -version -o /tmp/cc9wZoW3.s
GNU C++17 () version 11.1.0 (riscv64-unknown-elf)
	compiled by GNU C version 10.2.1 20210110, GMP version 6.2.1, MPFR version 4.1.0, MPC version 1.2.0, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/include/c++/11.1.0/riscv64-unknown-elf/rv64imafdc/lp64d"
ignoring nonexistent directory "/opt/riscv/riscv64-unknown-elf/usr/local/include"
ignoring duplicate directory "/opt/riscv/riscv64-unknown-elf/include"
#include "..." search starts here:
#include <...> search starts here:
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/include/c++/11.1.0
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/include/c++/11.1.0/backward
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/include
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/include-fixed
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/include
End of search list.
GNU C++17 () version 11.1.0 (riscv64-unknown-elf)
	compiled by GNU C version 10.2.1 20210110, GMP version 6.2.1, MPFR version 4.1.0, MPC version 1.2.0, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: 612ff88657548a42de426f214474ca73
In file included from hello.cpp:1:
/opt/riscv/riscv64-unknown-elf/include/c++/11.1.0/string:38:10: fatal error: bits/c++config.h: No such file or directory
   38 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```
```
root@8059ca370f34:/programs# riscv64-unknown-elf-g++ -march=rv64imafdc hello.cpp -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-g++
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv64-unknown-elf/11.1.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion= --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././riscv-gcc --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket --with-isa-spec=2.2 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medany'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 11.1.0 () 
COLLECT_GCC_OPTIONS='-march=rv64imafdc' '-v' '-mtune=rocket' '-mabi=lp64d' '-march=rv64imafdc' '-dumpdir' 'a-'
 /opt/riscv/libexec/gcc/riscv64-unknown-elf/11.1.0/cc1plus -quiet -v hello.cpp -quiet -dumpdir a- -dumpbase hello.cpp -dumpbase-ext .cpp -march=rv64imafdc -mtune=rocket -mabi=lp64d -march=rv64imafdc -version -o /tmp/ccv4RFLa.s
GNU C++17 () version 11.1.0 (riscv64-unknown-elf)
	compiled by GNU C version 10.2.1 20210110, GMP version 6.2.1, MPFR version 4.1.0, MPC version 1.2.0, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/opt/riscv/riscv64-unknown-elf/usr/local/include"
ignoring duplicate directory "/opt/riscv/riscv64-unknown-elf/include"
#include "..." search starts here:
#include <...> search starts here:
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/include/c++/11.1.0
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/include/c++/11.1.0/riscv64-unknown-elf
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/include/c++/11.1.0/backward
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/include
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/include-fixed
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/include
End of search list.
GNU C++17 () version 11.1.0 (riscv64-unknown-elf)
	compiled by GNU C version 10.2.1 20210110, GMP version 6.2.1, MPFR version 4.1.0, MPC version 1.2.0, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: 612ff88657548a42de426f214474ca73
COLLECT_GCC_OPTIONS='-march=rv64imafdc' '-v' '-mtune=rocket' '-mabi=lp64d' '-march=rv64imafdc' '-dumpdir' 'a-'
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/bin/as -v --traditional-format -march=rv64imafdc -march=rv64imafdc -mabi=lp64d -o /tmp/ccODMtGa.o /tmp/ccv4RFLa.s
GNU assembler version 2.38 (riscv64-unknown-elf) using BFD version (GNU Binutils) 2.38
COMPILER_PATH=/opt/riscv/libexec/gcc/riscv64-unknown-elf/11.1.0/:/opt/riscv/libexec/gcc/riscv64-unknown-elf/11.1.0/:/opt/riscv/libexec/gcc/riscv64-unknown-elf/:/opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/:/opt/riscv/lib/gcc/riscv64-unknown-elf/:/opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/bin/
LIBRARY_PATH=/opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/:/opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/lib/:/opt/riscv/riscv64-unknown-elf/lib/
COLLECT_GCC_OPTIONS='-march=rv64imafdc' '-v' '-mtune=rocket' '-mabi=lp64d' '-march=rv64imafdc' '-dumpdir' 'a.'
 /opt/riscv/libexec/gcc/riscv64-unknown-elf/11.1.0/collect2 -plugin /opt/riscv/libexec/gcc/riscv64-unknown-elf/11.1.0/liblto_plugin.so -plugin-opt=/opt/riscv/libexec/gcc/riscv64-unknown-elf/11.1.0/lto-wrapper -plugin-opt=-fresolution=/tmp/ccBp9ZVe.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgloss -plugin-opt=-pass-through=-lgcc --sysroot=/opt/riscv/riscv64-unknown-elf -melf64lriscv /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/lib/crt0.o /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/crtbegin.o -L/opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0 -L/opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/lib -L/opt/riscv/riscv64-unknown-elf/lib /tmp/ccODMtGa.o -lstdc++ -lm -lgcc --start-group -lc -lgloss --end-group -lgcc /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/crtend.o
COLLECT_GCC_OPTIONS='-march=rv64imafdc' '-v' '-mtune=rocket' '-mabi=lp64d' '-march=rv64imafdc' '-dumpdir' 'a.'
```

kito-cheng, 2022-07-21:
They are actually different architecture configure according the ISA spec definition rv64gc is rv64imafdc_zicsr_zifencei and rv64imafdc is rv64imafdc_zicsr...but I can understand it's annoying, I guess we should have a smarter multi-lib select mechanism.

ilg-ul, 2022-07-28:
I also encountered a similar problem, `rv64gc_zicsr` failed, but `rv64imafdc_zicsr` was accepted. A more tolerant multi-lib select mechanism would be great.

cmuellner, 2022-07-28:
This is a duplicate of https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1038