riscvarchive/riscv-gcc #346: hidden symbol `__dso_handle' isn't defined

ilg-ul, 2022-07-22:
This is a very old problem that plagued GCC. I remember seeing this on old versions of `arm-none-eabi-gcc` too, but apparently recent versions were fixed.

```
Building target: rv1.elf
Invoking: GNU RISC-V Cross C++ Linker
riscv-none-elf-g++ -march=rv32imac -mabi=ilp32 -mcmodel=medany -msmall-data-limit=8 -mno-save-restore -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -fno-move-loop-invariants -Wall -Wextra -g3 -T "/Users/ilg/My Files/WKS Projects/micro-os-plus.github/develop/riscv-workspace/rv1/xpacks/micro-os-plus-devices-qemu-riscv/linker-scripts/mem-virt-rv32gc.ld" -T "/Users/ilg/My Files/WKS Projects/micro-os-plus.github/develop/riscv-workspace/rv1/xpacks/micro-os-plus-architecture-riscv/linker-scripts/sections-ram.ld" -nostartfiles -L"../ldscripts" -Wl,-Map,"rv1.map" -o "rv1.elf"  ./xpacks/micro-os-plus-startup/src/_sbrk.o ./xpacks/micro-os-plus-startup/src/exit.o ./xpacks/micro-os-plus-startup/src/startup.o  ./xpacks/micro-os-plus-semihosting/src/semihosting-startup.o ./xpacks/micro-os-plus-semihosting/src/semihosting-syscalls.o ./xpacks/micro-os-plus-semihosting/src/semihosting-trace.o  ./xpacks/micro-os-plus-diag-trace/src/trace.o  ./xpacks/micro-os-plus-devices-qemu-riscv/src/reset-entry.o  ./src/main.o   
/Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.1.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/12.1.0/../../../../riscv-none-elf/bin/ld: /Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.1.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/12.1.0/../../../../riscv-none-elf/lib/libstdc++.a(cxx11-ios_failure.o): in function `_GLOBAL__sub_I__ZSt17iostream_categoryv':
(.text.startup._GLOBAL__sub_I__ZSt17iostream_categoryv+0x0): undefined reference to `__dso_handle'
/Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.1.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/12.1.0/../../../../riscv-none-elf/bin/ld: /Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.1.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/12.1.0/../../../../riscv-none-elf/lib/libstdc++.a(system_error.o): in function `.L0 ':
(.text.startup._GLOBAL__sub_I__ZSt20__throw_system_errori+0x2): undefined reference to `__dso_handle'
/Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.1.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/12.1.0/../../../../riscv-none-elf/bin/ld: (.text.startup._GLOBAL__sub_I__ZSt20__throw_system_errori+0x22): undefined reference to `__dso_handle'
/Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.1.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/12.1.0/../../../../riscv-none-elf/bin/ld: rv1.elf: hidden symbol `__dso_handle' isn't defined
/Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.1.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/12.1.0/../../../../riscv-none-elf/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
```

The symbol is not missing, it is hidden, and the linker seems not able to identify it.

One possible workaround is to define a weak such symbol:

```
void* __dso_handle __attribute__ ((weak));
```

It would be nice to not have to apply such hacks when compiling portable sources with the RISC-V compiler.

kito-cheng, 2022-07-22:
https://stackoverflow.com/questions/34308720/where-is-dso-handle-defined

That’s not responsibility of linker, you need to defined that in your crt stuffs if you don’t use default crt stuffs( -nostartfiles)

ilg-ul, 2022-07-22:
That's debatable, I checked the newlib sources and for arm & aarch64 I did not find this symbol in the startup code, and with the arm toolchains I do not get this error. However, the symbol is defined for some architectures (but not for RISC-V).

BTW, the post you are referring is 6 years old, and in the past the arm toolchain also had this problem.

kito-cheng, 2022-07-23:
`__dso_handle` defined in crtbegin.o which is provide by GCC/libgcc rather than newlib.

https://github.com/gcc-mirror/gcc/blob/master/libgcc/crtstuff.c#L338

ilg-ul, 2022-07-23:
You are right, the symbol is indeed defined in `crtbegin.o`, which is compiled from `crtstuff.c` in the second GCC build step.

When using `-nostartfiles` this file is not linked, and this symbol must be provided by the custom startup.

Thank you, I'll close this issue.

closed 2022-07-23 by ilg-ul