riscvarchive/riscv-gcc #349: -flto -> relocation truncated to fit: R_RISCV_GPREL_I against undefined symbol `_write'

ilg-ul, 2022-07-28:
I'm running some semihosting tests on QEMU, and, in a functional configuration, when I enabled `-flto` I got this relocation error:

```
: && /private/var/folders/gr/13tt3vcd7m1gnbhwtkmf5cnw0000gn/T/hello-world-qemu-template/005-riscv-rv64imafdc-cmake-c/xpacks/.bin/riscv-none-elf-g++ -O3 -DNDEBUG -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -fdiagnostics-color=always -march=rv64imafdc_zicsr -mabi=lp64d -mcmodel=medany -msmall-data-limit=8 -Werror -flto -fno-threadsafe-statics -nostartfiles -Wl,--gc-sections -Wl,-Map,platform-bin/hello-world-map.txt -T/private/var/folders/gr/13tt3vcd7m1gnbhwtkmf5cnw0000gn/T/hello-world-qemu-template/005-riscv-rv64imafdc-cmake-c/xpacks/micro-os-plus-devices-qemu-riscv/linker-scripts/mem-virt-rv64.ld -T/private/var/folders/gr/13tt3vcd7m1gnbhwtkmf5cnw0000gn/T/hello-world-qemu-template/005-riscv-rv64imafdc-cmake-c/xpacks/micro-os-plus-architecture-riscv/linker-scripts/sections-ram.ld platform-bin/CMakeFiles/hello-world.dir/__/src/main.c.obj platform-bin/CMakeFiles/hello-world.dir/__/xpacks/micro-os-plus-diag-trace/src/trace.cpp.obj platform-bin/CMakeFiles/hello-world.dir/__/xpacks/micro-os-plus-devices-qemu-riscv/src/reset-entry.S.obj platform-bin/CMakeFiles/hello-world.dir/__/xpacks/micro-os-plus-semihosting/src/semihosting-startup.cpp.obj platform-bin/CMakeFiles/hello-world.dir/__/xpacks/micro-os-plus-semihosting/src/semihosting-syscalls.cpp.obj platform-bin/CMakeFiles/hello-world.dir/__/xpacks/micro-os-plus-semihosting/src/semihosting-trace.cpp.obj platform-bin/CMakeFiles/hello-world.dir/__/xpacks/micro-os-plus-startup/src/startup.cpp.obj platform-bin/CMakeFiles/hello-world.dir/__/xpacks/micro-os-plus-startup/src/exit.c.obj platform-bin/CMakeFiles/hello-world.dir/__/xpacks/micro-os-plus-startup/src/_sbrk.c.obj -o platform-bin/hello-world.elf   && cd /private/var/folders/gr/13tt3vcd7m1gnbhwtkmf5cnw0000gn/T/hello-world-qemu-template/005-riscv-rv64imafdc-cmake-c/build/qemu-riscv-rv64imafdc-cmake-release/platform-bin && riscv-none-elf-size --format=berkeley /private/var/folders/gr/13tt3vcd7m1gnbhwtkmf5cnw0000gn/T/hello-world-qemu-template/005-riscv-rv64imafdc-cmake-c/build/qemu-riscv-rv64imafdc-cmake-release/platform-bin/hello-world.elf
/Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.1.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/12.1.0/../../../../riscv-none-elf/bin/ld: /Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.1.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/12.1.0/../../../../riscv-none-elf/lib/rv64imafdc_zicsr/lp64d/libc.a(lib_a-writer.o): in function `.L0 ':
(.text._write_r+0x18): undefined reference to `_write'
(.text._write_r+0x18): relocation truncated to fit: R_RISCV_GPREL_I against undefined symbol `_write'
/Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.1.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/12.1.0/../../../../riscv-none-elf/bin/ld: /Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.1.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/12.1.0/../../../../riscv-none-elf/lib/rv64imafdc_zicsr/lp64d/libc.a(lib_a-fstatr.o): in function `.L0 ':
(.text._fstat_r+0x16): undefined reference to `_fstat'
(.text._fstat_r+0x16): relocation truncated to fit: R_RISCV_GPREL_I against undefined symbol `_fstat'
/Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.1.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/12.1.0/../../../../riscv-none-elf/bin/ld: /Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.1.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/12.1.0/../../../../riscv-none-elf/lib/rv64imafdc_zicsr/lp64d/libc.a(lib_a-isattyr.o): in function `.L0 ':
(.text._isatty_r+0x12): undefined reference to `_isatty'
(.text._isatty_r+0x12): relocation truncated to fit: R_RISCV_GPREL_I against undefined symbol `_isatty'
/Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.1.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/12.1.0/../../../../riscv-none-elf/bin/ld: /Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.1.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/12.1.0/../../../../riscv-none-elf/lib/rv64imafdc_zicsr/lp64d/libc.a(lib_a-lseekr.o): in function `.L0 ':
(.text._lseek_r+0x18): undefined reference to `_lseek'
(.text._lseek_r+0x18): relocation truncated to fit: R_RISCV_GPREL_I against undefined symbol `_lseek'
/Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.1.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/12.1.0/../../../../riscv-none-elf/bin/ld: /Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.1.0-2.1/.content/bin/../lib/gcc/riscv-none-elf/12.1.0/../../../../riscv-none-elf/lib/rv64imafdc_zicsr/lp64d/libc.a(lib_a-readr.o): in function `.L0 ':
(.text._read_r+0x18): undefined reference to `_read'
(.text._read_r+0x18): relocation truncated to fit: R_RISCV_GPREL_I against undefined symbol `_read'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```
The application uses a local semihosting implementation, thus the need for local `_write()` & Co functions.

The problem occurs with both 32/64-bit builds. I tried to remove `-Wl,--gc-sections`, but it did not help.

A check with `nm` showed the symbols to be there:

```
% /Users/ilg/Library/xPacks/@xpack-dev-tools/riscv-none-elf-gcc/12.1.0-2.1/.content/bin/riscv-none-elf-nm /private/var/folders/gr/13tt3vcd7m1gnbhwtkmf5cnw0000gn/T/hello-world-qemu-template/005-riscv-rv64imafdc-cmake-c/build/qemu-riscv-rv64imafdc-cmake-release/platform-bin/CMakeFiles/hello-world.dir/__/xpacks/micro-os-plus-semihosting/src/semihosting-syscalls.cpp.obj
         U __errno
00000000 T _clock
00000000 T _close
00000000 T _execve
00000000 T _fork
00000000 T _fstat
00000000 T _ftime
00000000 T _getpid
00000000 T _gettimeofday
00000000 T _isatty
00000000 T _kill
00000000 T _link
00000000 T _lseek
00000000 T _open
00000000 T _read
00000000 T _rename
00000000 T _stat
00000000 T _system
00000000 T _times
00000000 T _unlink
00000000 T _wait
00000000 T _write
         U clock
         U close
00000000 T initialise_monitor_handles
         U open
         U stat
ilg@wksi ~ % 
```

I saw several similar messages on the net, but none were related to LTO. 

For now I'll keep `-flto` disabled, but it would be nice to enable it at some point.

kito-cheng, 2022-07-28:
Sounds weird, R_RISCV_GPREL_I is binutils internal relocation type which used during linker relaxation, so that sound something wrong in binutils, do you mind have more detail step for how to reproduce that and the version of your toolchian?

ilg-ul, 2022-07-28:
Yes, it is weird, and it might very well be a problem in binutils, as it might be a problem in my code, or in my build of the toolchain (https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases/tag/v12.1.0-2).

I'm currently almost ready with a template project that generates semihosted applications for various platforms (Arm and RISC-V) running on QEMU. If in the end the problem persists, I'll pass you details on how to use the template to generate a project that fails the build.

ilg-ul, 2022-07-29:
I don't have a diagnosis of the problem, but I have some more details; the good news is that the issue is not specific to RISC-V, but also affects `aarch64-none-elf-gcc` and `arm-none-eabi-gcc` (for aarch32).

Thus, from the dual error I would retain the _undefined reference_ and dismiss the _relocation truncated_, as a colateral damage. The reasoning is that the other toolchains issue similar _undefined reference_ messages.

My guess is that the problem is caused by an issue in the LTO logic, which ignores some symbols defined in the application. 

It would be interesting to find out why only these symbols are affected. It might be caused by some weak definitions, or other unusual configurations.

The issue affects only the LTO logic, the regular linker has no problem with these symbols, on all toolchains.

You can create test projects for all these architectures using the template project, and enable the `-flto` option, currently commented out:

- <https://github.com/micro-os-plus/hello-world-qemu-template-xpack/tree/xpack-develop>