riscvarchive/riscv-gcc #350: compile problem using --with-arch=rv64gcv_zfh

zqcawzs, 2022-08-04:
git submodule status:
 9826b03b747b841f5fc6de2054bf1ef3f5c4bdf3 glibc (glibc-2.33)
 85e0e3519655220688e757b9d5bfd314923548bd musl (v1.2.2)
 415fdd4279b85eeec9d54775ce13c5c412451e08 newlib (newlib-4.1.0)
 553032db17440f8de011390e5a1cfddd13751b0b qemu (v5.2.0)
 20756b0fbe065a84710aa38f2457563b57546440 riscv-binutils (heads/riscv-binutils-2.38)
 4ea498a8e1fafeb568530d84db1880066478c86b riscv-dejagnu (heads/riscv-dejagnu-1.6)
+edffbea0c0c4175ac26599b9da02df92b9c46866 riscv-gcc (heads/riscv-gcc-rvv-next)
 5da071ef0965b8054310d8dde9975037b0467311 riscv-gdb (remotes/origin/fsf-gdb-10.1-with-sim)

Facing many errors like the next:
rv64imafdcv_zfh_zve32f_zve32x_zve64d_zve64f_zve64x_zvl128b_zvl32b_zvl64b: unknown prefixed ISA extension `zfh'

zhongjuzhe, 2022-08-04:
You are using the wrong binutils which doesn't support 'zfh' extension. I suggest you use the latest upstream binutils. Or you can build toolchain with --with-arch=rv64gcv if you don't need FP16.

zqcawzs, 2022-08-04:
> You are using the wrong binutils which doesn't support 'zfh' extension. I suggest you use the latest upstream binutils. Or you can build toolchain with --with-arch=rv64gcv if you don't need FP16.

According to https://github.com/riscv-collab/riscv-binutils-gdb/issues/269, the zfh is not supported by binutils, am I right?

zhongjuzhe, 2022-08-04:
Yes, binutils-2.38 doesn't support ZFH. But the latest upstream binutils support ZFH.

zhongjuzhe, 2022-08-08:
@kito-cheng  Hi, Can you update binutils to upstream version in rvv-next. It seems that the current binutils doesn't support ZFH. Some newbie may not know how to assemble rvv toolchain submodule. Thanks.

cmuellner, 2022-08-08:
PR has been created: https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1108

zqcawzs, 2022-08-19:
> Yes, binutils-2.38 doesn't support ZFH. But the latest upstream binutils support ZFH.

Sorry for the late reply. Thanks for the help, the toolchain can be compiled using binutils-2.39.
One more question. Does this gcc version support all RVV ISA, including all kinds for data types?

zhongjuzhe, 2022-08-19:
> 

Sure.

zhongjuzhe, 2022-08-19:
> > Yes, binutils-2.38 doesn't support ZFH. But the latest upstream binutils support ZFH.
> 
> Sorry for the late reply. Thanks for the help, the toolchain can be compiled using binutils-2.39. One more question. Does this gcc version support all RVV ISA, including all kinds for data types?

YES.

zqcawzs, 2022-08-19:
> > > Yes, binutils-2.38 doesn't support ZFH. But the latest upstream binutils support ZFH.
> > 
> > 
> > Sorry for the late reply. Thanks for the help, the toolchain can be compiled using binutils-2.39. One more question. Does this gcc version support all RVV ISA, including all kinds for data types?
> 
> YES.

Excellent

closed 2022-08-19 by zqcawzs