riscvarchive/riscv-gcc #351: riscv64-unknown-linux-gnu/bin/as: unrecognized option '--64' from g++ during riscv toolchain build

Omar-M-Yehia, 2022-08-09:
during running make command I got the following issue 

TARGET_CPU_DEFAULT="" \
HEADERS="auto-host.h ansidecl.h" DEFINES="" \
/bin/sh ../../../riscv-gcc/gcc/mkconfig.sh config.h
TARGET_CPU_DEFAULT="" \
HEADERS="options.h insn-constants.h config/elfos.h config/gnu-user.h config/linux.h config/glibc-stdint.h config/riscv/riscv.h config/riscv/linux.h config/initfini-array.h defaults.h" DEFINES="LIBC_GLIBC=1 LIBC_UCLIBC=2 LIBC_BIONIC=3 LIBC_MUSL=4 DEFAULT_LIBC=LIBC_GLIBC ANDROID_DEFAULT=0 TARGET_DEFAULT_ISA_SPEC=ISA_SPEC_CLASS_2P2 TARGET_RISCV_ATTRIBUTE=0 TARGET_RISCV_DEFAULT_ARCH=rv64imafdc TARGET_RISCV_DEFAULT_ABI=lp64d" \
/bin/sh ../../../riscv-gcc/gcc/mkconfig.sh tm.h 
TARGET_CPU_DEFAULT="" \
HEADERS="config/riscv/riscv-protos.h config/linux-protos.h tm-preds.h" DEFINES="" \
/bin/sh ../../../riscv-gcc/gcc/mkconfig.sh tm_p.h
TARGET_CPU_DEFAULT="" \
HEADERS="auto-host.h ansidecl.h" DEFINES="" \
/bin/sh ../../../riscv-gcc/gcc/mkconfig.sh bconfig.h
g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -DGENERATOR_FILE -fno-PIE -I. -Ibuild -I../../../riscv-gcc/gcc -I../../../riscv-gcc/gcc/build -I../../../riscv-gcc/gcc/../include  -I../../../riscv-gcc/gcc/../libcpp/include  \
        -o build/genmodes.o ../../../riscv-gcc/gcc/genmodes.c
/home/omyehia/RISCV/riscv64-unknown-linux-gnu/bin/as: unrecognized option '--64'
make[2]: *** [build/genmodes.o] Error 1
 
I tried to use g++ 9.2.0 instead of g++ 4.8.5 , but this didn't solve the issue

kito-cheng, 2022-08-09:
I guess your native(x86) GCC invoked the wrong assembler, you should add the location of `riscv64-unknown-linux-gnu-as` into PATH instead of adding `/home/omyehia/RISCV/riscv64-unknown-linux-gnu/bin/` to PATH (I guess might be `/home/omyehia/RISCV/bin`).

Omar-M-Yehia, 2022-08-09:
I have already added /home/omyehia/RISCV/bin to PATH , the issue is fixed for me when I removed "." from  PATH

closed 2022-08-09 by Omar-M-Yehia