riscvarchive/riscv-gcc #353: How to update my riscv-gnu-toolchain to enable RVV support

schwa1z, 2022-08-24:
Hello community!

For some reason I want to use GCC as the compiler to support riscv-vector extension compilation. Long ago, I have installed gcc and riscv-gcc. The following is the version information now:

when I type ```gcc --version```in terminal:
```
$gcc --version
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
and about riscv-gcc, the information in /riscv-gnu-toolchain/readme.md:
```
The branch info has recorded in `.gitmodules` file, which can set or update via
`git submodule add -b` or `git submodule set-branch`.

However the only way to check which branch are using is to check `.gitmodules`
file, here is the example for `riscv-gcc`, it using riscv-gcc-10.2.0 branch, so
it will has a section named `riscv-gcc` and has a field `branch` is
`riscv-gcc-10.2.0`.

[submodule "riscv-gcc"]
        path = riscv-gcc
        url = ../riscv-gcc.git
        branch = riscv-gcc-10.2.0
```
since I'm confused about the relationship between gcc and riscv-gcc, I put all the information here. RVV extension is not supported in this version.

Now I want to update the version of my toolchain to enable RVV extension support. What should I do and what command should I type?

Thank you very much!

zhongjuzhe, 2022-08-24:
You should checkout to branch 'riscv-gcc-rvv-next'

schwa1z, 2022-08-24:
Thank you for your reply! @zhongjuzhe 
I have switch the branch to rvv-next and build the toolchain successfully. But there is another question.
The riscv core I used need the follow files:
```
    RISCV_CC            := riscv32-unknown-elf-gcc
    RISCV_DUMP          := riscv32-unknown-elf-objdump
    RISCV_OBCP          := riscv32-unknown-elf-objcopy
    ARCH                := rv32imv
    LD_SCRIPT           := $(SW_DIR)/link.ld
``` 
While in install_path/bin/:
[image]

It looks like I don't build up the toolchain correctly, because the core need 32bit files but I have 64bit now.

After that what should I do to satisfy the need of the riscv core?

zhongjuzhe, 2022-08-24:
You should specify --arch=rv32gcv

schwa1z, 2022-08-24:
Thanks! I reconfigured the toolchain like this:
[image]

while the error occurs:
[image]

Do you have any idea?

pz9115, 2022-08-24:
I think you should try to use --with-arch=rv32gcv in configure.

schwa1z, 2022-08-24:
Hi! @pz9115 . 
As you can see, I actually used --with-arch=rv32gcv in the first figure here. But there is an error in the second figure. Do you have any idea?

> Thanks! I reconfigured the toolchain like this: [image]
> 
> while the error occurs: [image]
> 
> Do you have any idea?

TommyMurphyTM1234, 2022-08-24:
Please note that `riscv64-unknown-elf` and `riscv32-unknown-elf` prefixed tools can both build 32 and 64 bit RISC-V software.

The `riscv64-unknown-elf` toolchain that you built first will work if your target is RV32 if you pass the appropriate `-march=` and `-mabi=` options when compiling software for your target.

You'll also need to specify `--enable-multilib` at configuration time before building the toolsn so that the toolchain includes RV32 multilibs and not just the default rv64gc libs.

If your makefile/build scripts reference `riscv32-unknown-elf` prefixed tools then just change them to reference `riscv64-unknown-elf` tools.

TommyMurphyTM1234, 2022-08-25:
Actually, on reflection, it's probably simpler for your purposes, if you're targeting rv32imv, to do...

```
./configure --with-arch=rv32imv --with-abi=ilp32
make
```

to build the toolchain just for your target architecture/abi.

schwa1z, 2022-08-25:
Thank you for your patient reply! @TommyMurphyTM1234 .I benefited a lot from your answer.
Actually, after reading your first answer, I have configured the toolchain like the second answer you provide.
But I encountered another problem, which is the riscv core couldn'd print any output information.
When I compile vector-mode C code(with RVV Intrinsic style). There is no error occur, but there is no output too.
However, when I compile normal C code(without RVV instruction), it works.

**I want to know, is the rvv-next branch support Intrinsic coding?**
Or I need to code in inline assembly style?

When I choose LLVM as the compiler(which is successful) to compile RVV Intrinsic C code, the output would be like:
[image]
When I choose GCC to compile RVV Intrinsic C code, the output would be like:
[image]
There is no output, which makes me confused. Could you help me?

zhongjuzhe, 2022-08-25:
> Thank you for your patient reply! @TommyMurphyTM1234 .I benefited a lot from your answer. Actually, after reading your first answer, I have configured the toolchain like the second answer you [...]

rvv-next support all RVV feature including intrinsic and auto-vectorization. Would you mind share your codes? And also you need to check whether you are using the latest branch "riscv-gcc-rvv-next" in riscv-gcc.

schwa1z, 2022-08-25:
Thank you @zhongjuzhe , no problem, here is the main code:
```
    for (int n = sizeof(greydata); n > 0; n -= vl, src_B += vl, src_G += vl, src_R += vl, dst += vl, src_p151 += vl) {

        vl            = vsetvl_e8m1(n);
        vuint16m2_t vec_dst = vle16_v_u16m2(dst, vl);

        vuint8m1_t vec_B = vle8_v_u8m1(src_B, vl);
        vuint8m1_t vec_G = vle8_v_u8m1(src_G, vl);
        vuint8m1_t vec_R = vle8_v_u8m1(src_R, vl);

        vuint8m1_t vec_p = vle8_v_u8m1(src_p151, vl);
 
        vec_dst           = vwmaccu_vx_u16m2(vec_dst, 28, vec_B, vl);
        vec_dst           = vwmaccu_vv_u16m2(vec_dst, vec_p, vec_G, vl);
        vec_dst           = vwmaccu_vx_u16m2(vec_dst, 77, vec_R, vl);

        
        vec_dst           = vsrl_vx_u16m2(vec_dst, 8, vl);
        vse16_v_u16m2(dst, vec_dst, vl);
    }
```
I don't know how to check the version of my toolchain. I download it yesterday. Could you tell me how to check the version?
I used this command to download:
```
git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-next
``` 
and this command to make:
```
./configure --prefix=$RISCV --with-arch=rv32imv --with-abi=ilp32 --enable-multilib
make
```

zhongjuzhe, 2022-08-25:
cd riscv-gcc && git log

zhongjuzhe, 2022-08-25:
> Thank you @zhongjuzhe , no problem, here is the main code:
>
> ```
>     for (int n = sizeof(greydata); n > 0; n -= vl, src_B += vl, src_G += vl, src_R += vl, dst += vl, src_p151 += vl) { [...]

Would you mind giving me the whole program? including the main function and data initilization.

zhongjuzhe, 2022-08-25:
No, plz git log in riscv-gnu-toolchain directory. Plz git log riscv-gcc directly. cd riscv-gcc and then git log.

schwa1z, 2022-08-25:
I'm sorry. Here it is:
[image]

zhongjuzhe, 2022-08-25:
> I'm sorry. Here it is: [image]

These codes are obsolete. Plz git pull in **riscv-gcc** and rebuild the toolchain.  The latest commit of risc-gcc-rvv-next should be **commit 53fcb21e7d050c07ce5179a9e80bd1ee2528fd62**. And try your program again.

schwa1z, 2022-08-26:
That's the problem! @zhongjuzhe  I pull the latest commit and compile the code successfully. 
All clear! Thank everyone in this post for your help! I'm very grateful!

closed 2022-08-26 by schwa1z

HamzaShabbir517, 2022-08-27:
@schwa1z Hope you are doing good. Can you please share the steps you follow to compile the Vector C code. As i am also trying to configure the toolchain for RV32IMFCV target. I have gone through the whole conversation and as mention i also try git log and it shows the same message as you post it here but when it try to run git pull it gives me an error so may be i am doing something wrong. Your steps may help me out. Thanks in advance

Just for reference sharing the snap 
[image]

schwa1z, 2022-08-28:
@HamzaShabbir517 Yes, I'm glad to share my method to configure the toolchain.
```
git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-next
```
then, in .../riscv-gnu-toolchain/, switch to 'riscv-gcc-rvv-next'branch, and pull the latest version
```
cd riscv-gcc
git checkout -b riscv-gcc-rvv-next
git pull
```
After that, configure the toolchain:(for you maybe --with-arch=rv32imfcv? I'm not sure), note that you need to configure the environment varible in bashrc. If you don't know how to configure the environment varible, tell me.
```
./configure --prefix=$RISCV --with-arch=rv32imv --with-abi=ilp32 --enable-multilib
```
and make:
```
sudo make
```

HamzaShabbir517, 2022-08-28:
@schwa1z Thanks i was able to build it. 
The another question i have is where can I find the definitions of functions of vector imstruction like you have used in your program.
And is there any way to tell the compiler not to generate specific instructions as we all know vector extension contain lots of instructions and i have implement only few of them.

schwa1z, 2022-08-28:
@HamzaShabbir517. There are two ways to call riscv vector instructions. 
One is to use intrinsic(which is a API), like my code above. You can refer to   [https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/master/rvv-intrinsic-rfc.md](url) and **_riscv-v-spec-1.0_** to write intrinsic code.
Another is to use inline assembly in C code. To be honest, I'm not good at this way. But you could find many reference on Internet.

I'm a little bit of confused about your last question. When you write specific Intrinsic/inline assembly in C code to call vector instruction, the compiler just compile the specific instruction you write, instead of translating other instructions.

HamzaShabbir517, 2022-08-28:
Below is the snap of the C code i write for simple vector addition. 
[image]

The assembly generated by the compiler include **vl1re32.v** instructions as well in some places as my hardware does not support this instruction so i want the compiler not to generate such instruction

schwa1z, 2022-08-28:
I don't find this instruction in **_riscv-v-spec-1.0_**. But it looks strange. I don't know how to prevent the compiler to generate specific instruction. Maybe you could open a new issue.

zhongjuzhe, 2022-08-28:
vl1re32 is whole register load which is a common use instruction.  You can find it inchapter 7.9---- Replied Message ***@***.***>Date08/28/2022 15:46 ***@***.***> ***@***.******@***.***>SubjectRe: [riscv-collab/riscv-gcc] How to update my riscv-gnu-toolchain to enable RVV support (Issue #353)
I don't find this instruction in riscv-v-spec-1.0. But it looks strange. I don't know how to prevent the compiler to generate specific instruction. Maybe you could open a new issue.

[
{
"target": "https://github.com/riscv-collab/riscv-gcc/issues/353#issuecomment-1229400350",
"url": "https://github.com/riscv-collab/riscv-gcc/issues/353#issuecomment-1229400350",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

TommyMurphyTM1234, 2022-08-28:
https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1123#issuecomment-1229433844

zhongjuzhe, 2022-08-28:
> Below is the snap of the C code i write for simple vector addition. [image]
> 
> The assembly generated by the compiler include **vl1re32.v** instructions as well in some places as my hardware does not support this instruction so i want the compiler not to generate such instruction

I have tried your code using riscv64-unknow-elf-gcc -O3:
generated assmebly:
foo:
        beq     a0,zero,.L1
.L3:
        vsetvli a5,a0,e32,m1,ta,mu
        slli    a4,a5,2
        vle32.v v24,(a2)
        vle32.v v25,(a3)
        add     a2,a2,a4
        vadd.vv v24,v24,v25
        add     a3,a3,a4
        vse32.v v24,(a1)
        add     a1,a1,a4
        j       .L3
.L1:
        ret

**There is no vl1re32.v instructions** when I compile using O3. 
If you use O0 to compile, it may possible generate vl1re32.v because compiler
should use conservative way to compile the c codes.

Here is code-gen in LLVM:
https://godbolt.org/z/ecdfhreTW
GCC behaves same as LLVM.

HamzaShabbir517, 2022-08-28:
@zhongjuzhe Here is the assembly generated by GCC Compiler i also used -03 flag in my compile command
[image]

HamzaShabbir517, 2022-08-28:
thanks everyone for helping me out. Really appreciate it.

zhongjuzhe, 2022-08-28:
It's wierld. I don't know how you get this. 
My compilation works well:
codes:
#include <riscv_vector.h>

void foo (size_t n, uint32_t a[n], const uint32_t b[n], const uint32_t c[n])
{
  while (n > 0)
    {
        size_t vl = vsetvl_e32m1 (n);
        vuint32m1_t vb = vle32_v_u32m1 (b, vl);
        vuint32m1_t vc = vle32_v_u32m1 (c, vl);
        vuint32m1_t va = vadd_vv_u32m1 (vb, vc, vl);
        vse32_v_u32m1 (a, va, vl);
        a += vl;
        b += vl;
        c += vl;
        n -+ vl;
    }
}

assembly:
foo:
        beq     a0,zero,.L1
.L3:
        vsetvli a5,a0,e32,m1,ta,mu
        slli    a4,a5,2
        vle32.v v24,(a2)
        vle32.v v25,(a3)
        add     a2,a2,a4
        vadd.vv v24,v24,v25
        add     a3,a3,a4
        vse32.v v24,(a1)
        add     a1,a1,a4
        j       .L3
.L1:
        ret

***@***.***
 
From: HamzaShabbir517
Date: 2022-08-28 21:06
To: riscv-collab/riscv-gcc
CC: zhongjuzhe; Mention
Subject: Re: [riscv-collab/riscv-gcc] How to update my riscv-gnu-toolchain to enable RVV support (Issue #353)
@zhongjuzhe Here is the assembly generated by GCC Compiler i also used -03 flag in my compile command

HamzaShabbir517, 2022-08-28:
@zhongjuzhe Can you try this code as well at your side.
[image]

I am getting this assembly
ee000000 <vector_add>:
ee000000:	c115                	beqz	a0,ee000024 <vector_add+0x24>
ee000002:	050577d7          	vsetvli	a5,a0,e32,m1,ta,mu
ee000006:	00279713          	slli	a4,a5,0x2
ee00000a:	02066c07          	vle32.v	v24,(a2)
ee00000e:	0206ec87          	vle32.v	v25,(a3)
ee000012:	8d1d                	sub	a0,a0,a5
ee000014:	038c8c57          	vadd.vv	v24,v24,v25
ee000018:	0205ec27          	vse32.v	v24,(a1)
ee00001c:	963a                	add	a2,a2,a4
ee00001e:	95ba                	add	a1,a1,a4
ee000020:	96ba                	add	a3,a3,a4
ee000022:	f165                	bnez	a0,ee000002 <vector_add+0x2>
ee000024:	8082                	ret

Disassembly of section .text.startup:

ee000026 <main>:
ee000026:	f0040737          	lui	a4,0xf0040
ee00002a:	711d                	addi	sp,sp,-96
ee00002c:	00070613          	mv	a2,a4
ee000030:	02000793          	li	a5,32
ee000034:	868a                	mv	a3,sp
ee000036:	00070713          	mv	a4,a4
ee00003a:	0c37f5d7          	vsetvli	a1,a5,e8,m8,ta,ma
ee00003e:	02070c07          	vle8.v	v24,(a4)
ee000042:	8f8d                	sub	a5,a5,a1
ee000044:	02068c27          	vse8.v	v24,(a3)
ee000048:	972e                	add	a4,a4,a1
ee00004a:	96ae                	add	a3,a3,a1
ee00004c:	f7fd                	bnez	a5,ee00003a <main+0x14>
ee00004e:	02000793          	li	a5,32
ee000052:	1018                	addi	a4,sp,32
ee000054:	0c37f6d7          	vsetvli	a3,a5,e8,m8,ta,ma
ee000058:	02060c07          	vle8.v	v24,(a2)
ee00005c:	8f95                	sub	a5,a5,a3
ee00005e:	02070c27          	vse8.v	v24,(a4)
ee000062:	9636                	add	a2,a2,a3
ee000064:	9736                	add	a4,a4,a3
ee000066:	f7fd                	bnez	a5,ee000054 <main+0x2e>
ee000068:	0094                	addi	a3,sp,64
ee00006a:	1008                	addi	a0,sp,32
ee00006c:	880a                	mv	a6,sp
ee00006e:	4721                	li	a4,8
ee000070:	050777d7          	vsetvli	a5,a4,e32,m1,ta,mu
ee000074:	00279613          	slli	a2,a5,0x2
ee000078:	02086c07          	vle32.v	v24,(a6)
ee00007c:	02056c87          	vle32.v	v25,(a0)
ee000080:	8f1d                	sub	a4,a4,a5
ee000082:	038c8c57          	vadd.vv	v24,v24,v25
ee000086:	0206ec27          	vse32.v	v24,(a3)
ee00008a:	9832                	add	a6,a6,a2
ee00008c:	96b2                	add	a3,a3,a2
ee00008e:	9532                	add	a0,a0,a2
ee000090:	f365                	bnez	a4,ee000070 <main+0x4a>
ee000092:	6125                	addi	sp,sp,96
ee000094:	8082                	ret

TommyMurphyTM1234, 2022-08-28:
> @zhongjuzhe Can you try this code as well at your side. [image]

You really should post the actual code and not a screenshot expecting the other person to manually type the code in.

zhongjuzhe, 2022-08-28:
> @zhongjuzhe Can you try this code as well at your side. #include <riscv_vector.h> void vector_add(size_t n, int32_t a[n], const int32_t b[n], const int32_t c[n]) {
> 
> ```
> while (n>0) 
> {
> size_t vl = vsetvl_e32m1(n);
> vint32m1_t vb = vle32_v_i32m1(b,vl);
> vint32m1_t vc = vle32_v_i32m1(c,vl);
> vint32m1_t va = vadd_vv_i32m1(vb,vc,vl);
> vse32_v_i32m1(a, va,vl);
> a += vl;
> b += vl;
> c += vl;
> n -= vl;
> }
> ```
> 
> }
> 
> void main() { size_t n = 8; int32_t A[] = {1,2,3,4,5,6,7,8}; int32_t B[] = {1,2,3,4,5,6,7,8}; int32_t C[8];
> 
> ```
> vector_add(n,C,A,B);
> ```
> 
> }
> 
> I am getting this assembly ee000000 <vector_add>: ee000000: c115 beqz a0,ee000024 <vector_add+0x24> ee000002: 050577d7 vsetvli a5,a0,e32,m1,ta,mu ee000006: 00279713 slli a4,a5,0x2 ee00000a: 02066c07 vle32.v v24,(a2) ee00000e: 0206ec87 vle32.v v25,(a3) ee000012: 8d1d sub a0,a0,a5 ee000014: 038c8c57 vadd.vv v24,v24,v25 ee000018: 0205ec27 vse32.v v24,(a1) ee00001c: 963a add a2,a2,a4 ee00001e: 95ba add a1,a1,a4 ee000020: 96ba add a3,a3,a4 ee000022: f165 bnez a0,ee000002 <vector_add+0x2> ee000024: 8082 ret
> 
> Disassembly of section .text.startup:
> 
> ee000026
> 
> :
> ee000026: f0040737 lui a4,0xf0040
> ee00002a: 711d addi sp,sp,-96
> ee00002c: 00070613 mv a2,a4
> ee000030: 02000793 li a5,32
> ee000034: 868a mv a3,sp
> ee000036: 00070713 mv a4,a4
> ee00003a: 0c37f5d7 vsetvli a1,a5,e8,m8,ta,ma
> ee00003e: 02070c07 vle8.v v24,(a4)
> ee000042: 8f8d sub a5,a5,a1
> ee000044: 02068c27 vse8.v v24,(a3)
> ee000048: 972e add a4,a4,a1
> ee00004a: 96ae add a3,a3,a1
> ee00004c: f7fd bnez a5,ee00003a <main+0x14>
> ee00004e: 02000793 li a5,32
> ee000052: 1018 addi a4,sp,32
> ee000054: 0c37f6d7 vsetvli a3,a5,e8,m8,ta,ma
> ee000058: 02060c07 vle8.v v24,(a2)
> ee00005c: 8f95 sub a5,a5,a3
> ee00005e: 02070c27 vse8.v v24,(a4)
> ee000062: 9636 add a2,a2,a3
> ee000064: 9736 add a4,a4,a3
> ee000066: f7fd bnez a5,ee000054 <main+0x2e>
> ee000068: 0094 addi a3,sp,64
> ee00006a: 1008 addi a0,sp,32
> ee00006c: 880a mv a6,sp
> ee00006e: 4721 li a4,8
> ee000070: 050777d7 vsetvli a5,a4,e32,m1,ta,mu
> ee000074: 00279613 slli a2,a5,0x2
> ee000078: 02086c07 vle32.v v24,(a6)
> ee00007c: 02056c87 vle32.v v25,(a0)
> ee000080: 8f1d sub a4,a4,a5
> ee000082: 038c8c57 vadd.vv v24,v24,v25
> ee000086: 0206ec27 vse32.v v24,(a3)
> ee00008a: 9832 add a6,a6,a2
> ee00008c: 96b2 add a3,a3,a2
> ee00008e: 9532 add a0,a0,a2
> ee000090: f365 bnez a4,ee000070 <main+0x4a>
> ee000092: 6125 addi sp,sp,96
> ee000094: 8082 ret

You should use -fno-tree-vectorize to disable auto-vectorization

HamzaShabbir517, 2022-08-29:
i think GCC does not support this flag as nothing is happening

zhongjuzhe, 2022-08-29:
> I have tried the flag as you mention but nothing happen got the same assembly as above. These are the CFlags i am using
> 
> TEST_CFLAGS = -march=rv32imfcv -mabi=ilp32f -O3 -fno-tree-vectorize

You said :
ee000000 <vector_add>:
ee000000: c115 beqz a0,ee000024 <vector_add+0x24>
ee000002: 050577d7 vsetvli a5,a0,e32,m1,ta,mu
ee000006: 00279713 slli a4,a5,0x2
ee00000a: 02066c07 vle32.v v24,(a2)
ee00000e: 0206ec87 vle32.v v25,(a3)
ee000012: 8d1d sub a0,a0,a5
ee000014: 038c8c57 vadd.vv v24,v24,v25
ee000018: 0205ec27 vse32.v v24,(a1)
ee00001c: 963a add a2,a2,a4
ee00001e: 95ba add a1,a1,a4
ee000020: 96ba add a3,a3,a4
ee000022: f165 bnez a0,ee000002 <vector_add+0x2>
ee000024: 8082 ret

This assembly is correct. What's the problem?

HamzaShabbir517, 2022-08-29:
The problem is in the main loop:
ee000026 <main>:
ee000026:	f0040737          	lui	a4,0xf0040
ee00002a:	7159                	addi	sp,sp,-112
ee00002c:	00070613          	mv	a2,a4
ee000030:	02400793          	li	a5,36
ee000034:	1034                	addi	a3,sp,40
ee000036:	00070713          	mv	a4,a4
ee00003a:	0c37f5d7          	vsetvli	a1,a5,e8,m8,ta,ma
ee00003e:	02070c07          	vle8.v	v24,(a4)
ee000042:	8f8d                	sub	a5,a5,a1
ee000044:	02068c27          	vse8.v	v24,(a3)
ee000048:	972e                	add	a4,a4,a1
ee00004a:	96ae                	add	a3,a3,a1
ee00004c:	f7fd                	bnez	a5,ee00003a <main+0x14>

as here u can see to load scalar data it is also using vector instruction i want to stop that

zhongjuzhe, 2022-08-29:
> 

You can use -Os instead of -O3. It will stop optimizing "memcpy" using vector instructions.

HamzaShabbir517, 2022-08-29:
I have already try that as whenever i use -Os it gives me the following error
[image]

zhongjuzhe, 2022-08-29:
> I have already try that as whenever i use -Os it gives me the following error [image]

This is the issue of **linker**. If you want to use -march=rv32imfcv -mabi=ilp32f.
You should build a rv32 toolchain which generate **riscv32-unknown-elf-gcc**.

Before that you could use the riscv64-unknown-elf-gcc -march=rv64imfcv -mabi=lp64f -Os to check whether my
suggestion is working for you for now.

TommyMurphyTM1234, 2022-08-29:
> I have already try that as whenever i use -Os it gives me the following error [image]

Looks like your toolchain doesn't have the multilib for `rc32imfcv/ilp32f`. If you built a multilib toolchain then note that that arch/abi is not in the default list of multilibs built.

https://github.com/gcc-mirror/gcc/blob/master/gcc/config/riscv/t-elf-multilib

You might be better off building a toolchain for your specific target as I suggested previously.

`./configure --with-arch=rv32imfcv --with-abi=ilp32f --prefix=...`

TommyMurphyTM1234, 2022-08-29:
> > I have already try that as whenever i use -Os it gives me the following error [image]
> 
> This is the issue of **linker**. If you want to use -march=rv32imfcv -mabi=ilp32f. You should build a rv32 toolchain which generate **riscv32-unknown-elf-gcc**.

Just being rv32 is not sufficient. It will need to be a toolchain that targets the specific arch/abi used here.

`./configure --with-arch=rv32imfcv --with-abi=ilp32f --prefix=...`

HamzaShabbir517, 2022-08-29:
I have build both the toolchains for 64 and 32 by using the following commands

for 64 bit
../configure --prefix=/home/hshabbir/local/riscvv09/gnu/ --with-arch=rv64imfcv --enable-multilib

for 32 bit
./configure --with-arch=rv32imfcv_zicsr_zifencei --with-abi=ilp32f --prefix=/home/hshabbir/local/rv32imfcv --enable-multilib 

But still will try to build a specific toolchain and will try to compile it again

TommyMurphyTM1234, 2022-08-29:
> for 64 bit ../configure --prefix=/home/hshabbir/local/riscvv09/gnu/ --with-arch=rv64imfcv --enable-multilib

You don't need this toolchain.

> for 32 bit ./configure --with-arch=rv32imfcv_zicsr_zifencei --with-abi=ilp32f --prefix=/home/hshabbir/local/rv32imfcv --enable-multilib

You don't need multilib.

I forgot about `_Zicsr_Zifencei` and am not sure that you need to specify it here, but if it works and doesn't cause any error then stick with it.

HamzaShabbir517, 2022-08-29:
ok i will try it and get back to you

HamzaShabbir517, 2022-08-29:
@TommyMurphyTM1234 @zhongjuzhe I have rebuild the toolchain by using the following command
./configure --with-arch=rv32imfcv--with-abi=ilp32f

Now when i compile the code by using following CFLAGS
TEST_CFLAGS = -march=rv32imfcv -mabi=ilp32f -Os

The code got compile by this have some memcy function in which there are lots of instructions
Disassembly of section .text:

ee000000 <vector_add>:
ee000000:	e111                	bnez	a0,ee000004 <vector_add+0x4>
ee000002:	8082                	ret
ee000004:	050577d7          	vsetvli	a5,a0,e32,m1,ta,mu
ee000008:	02066c07          	vle32.v	v24,(a2)
ee00000c:	0206ec87          	vle32.v	v25,(a3)
ee000010:	038c8c57          	vadd.vv	v24,v24,v25
ee000014:	0205ec27          	vse32.v	v24,(a1)
ee000018:	00279713          	slli	a4,a5,0x2
ee00001c:	95ba                	add	a1,a1,a4
ee00001e:	963a                	add	a2,a2,a4
ee000020:	96ba                	add	a3,a3,a4
ee000022:	8d1d                	sub	a0,a0,a5
ee000024:	bff1                	j	ee000000 <vector_add>

Disassembly of section .text.startup:

ee000026 <main>:
ee000026:	7159                	addi	sp,sp,-112
ee000028:	d4a2                	sw	s0,104(sp)
ee00002a:	f0040437          	lui	s0,0xf0040
ee00002e:	00040593          	mv	a1,s0
ee000032:	02000613          	li	a2,32
ee000036:	850a                	mv	a0,sp
ee000038:	d686                	sw	ra,108(sp)
ee00003a:	026000ef          	jal	ra,ee000060 <memcpy>
ee00003e:	00040593          	mv	a1,s0
ee000042:	02000613          	li	a2,32
ee000046:	1008                	addi	a0,sp,32
ee000048:	018000ef          	jal	ra,ee000060 <memcpy>
ee00004c:	1014                	addi	a3,sp,32
ee00004e:	860a                	mv	a2,sp
ee000050:	008c                	addi	a1,sp,64
ee000052:	4521                	li	a0,8
ee000054:	fadff0ef          	jal	ra,ee000000 <vector_add>
ee000058:	50b6                	lw	ra,108(sp)
ee00005a:	5426                	lw	s0,104(sp)
ee00005c:	6165                	addi	sp,sp,112
ee00005e:	8082                	ret

Disassembly of section .text.memcpy:

ee000060 <memcpy>:
ee000060:	00b547b3          	xor	a5,a0,a1
ee000064:	8b8d                	andi	a5,a5,3
ee000066:	00c508b3          	add	a7,a0,a2
ee00006a:	ebf9                	bnez	a5,ee000140 <memcpy+0xe0>
ee00006c:	478d                	li	a5,3
ee00006e:	0cc7f963          	bgeu	a5,a2,ee000140 <memcpy+0xe0>
ee000072:	00357793          	andi	a5,a0,3
ee000076:	872a                	mv	a4,a0
ee000078:	eba9                	bnez	a5,ee0000ca <memcpy+0x6a>
ee00007a:	ffc8f793          	andi	a5,a7,-4
ee00007e:	40e78633          	sub	a2,a5,a4
ee000082:	02000693          	li	a3,32
ee000086:	02000293          	li	t0,32
ee00008a:	06c6c363          	blt	a3,a2,ee0000f0 <memcpy+0x90>
ee00008e:	02f77b63          	bgeu	a4,a5,ee0000c4 <memcpy+0x64>
ee000092:	17fd                	addi	a5,a5,-1
ee000094:	8f99                	sub	a5,a5,a4
ee000096:	8389                	srli	a5,a5,0x2
[... 104 lines cut ...]
        sw      a7, 52(sp)
        sw      a3, 48(sp)
        sw      a4, 44(sp)
        sw      a5, 40(sp)
        sw      a1, 36(sp)
        sw      a2, 32(sp)
        addi    a1, sp, 32
        addi    a2, sp, 64
        mv      a3, sp
.LBB1_1:                                # =>This Inner Loop Header: Depth=1
        vsetvli a4, a0, e32, m1, ta, mu
        vle32.v v8, (a2)
        vle32.v v9, (a1)
        vadd.vv v8, v8, v9
        vse32.v v8, (a3)
        slli    a5, a4, 2
        add     a3, a3, a5
        add     a2, a2, a5
        sub     a0, a0, a4
        add     a1, a1, a5
        bnez    a0, .LBB1_1
        addi    sp, sp, 96
        ret

Now what should i do

HamzaShabbir517, 2022-08-29:
This is the make file command i am using to build the code

ifneq (,$(wildcard $(TEST_DIR)/$(TEST).makefile))
program.hex:
	@echo Building $(TEST) via $(TEST_DIR)/$(TEST).makefile
	$(MAKE) -f $(TEST_DIR)/$(TEST).makefile
else
program.hex: $(OFILES) $(LINK)
	@echo Building $(TEST)
	$(GCC_PREFIX)-gcc $(ABI) -Wl,-Map=$(TEST).map -lgcc -T$(LINK) -o $(TEST).exe $(OFILES) -nostartfiles -lm $(TEST_LIBS)
	$(GCC_PREFIX)-objcopy -O verilog  $(TEST).exe program.hex
	$(GCC_PREFIX)-objdump -S $(TEST).exe > $(TEST).dis
	@echo Completed building $(TEST)

%.o : %.c ${BUILD_DIR}/defines.h
	$(GCC_PREFIX)-gcc ${includes} ${TEST_CFLAGS} -DCOMPILER_FLAGS="\"${TEST_CFLAGS}\"" ${ABI} -nostdlib -c $< -o $@

TommyMurphyTM1234, 2022-08-29:
Your makefile snippet is impossible to read.
You should put it inside code tags.

HamzaShabbir517, 2022-08-29:
[image]

HamzaShabbir517, 2022-08-29:
@TommyMurphyTM1234 I have updated the code in the previous comment of mine

zhongjuzhe, 2022-08-29:
> @TommyMurphyTM1234 I have updated the code in the previous comment of mine

OK. I understand your problem now. You want to use scalar instructions directly to do the "memcpy".
In RVV, I use RVV instructions to do this.

So please go to this file: riscv-gcc/gcc/config/riscv/riscv.md.

search "cpymemsi".
You will find this:

(define_expand "cpymemsi"
  [(parallel [(set (match_operand:BLK 0 "general_operand")
		   (match_operand:BLK 1 "general_operand"))
	      (use (match_operand:SI 2 ""))
	      (use (match_operand:SI 3 "const_int_operand"))])]
  ""
{
  **if (riscv_vector_expand_block_move (operands))
    DONE;
  else** if (riscv_expand_block_move (operands[0], operands[1], operands[2]))
    DONE;
  else
    FAIL;
})

Remove the codes that I highlights.

zhongjuzhe, 2022-08-29:
> @TommyMurphyTM1234 I have updated the code in the previous comment of mine

 I understand your problem now. You want to use scalar instructions to do the memcpy. However, current rvv-next enable optimizing memcpy using RVV instructions. I will add a compile option to disable this today. After I have done this, I will tell you to pull the latest codes.

zhongjuzhe, 2022-08-30:
> @TommyMurphyTM1234 I have updated the code in the previous comment of mine
>
> I understand your problem now. You want to use scalar instructions to do the memcpy. However, current rvv-next enable [...]

Plz pull the latest codes, the latest code should be:
42df346446308053814cb61601a1470e26ab1cf7
Then rebuild the toolchain. use -O3 -fno-tree-vectorize to compile your program

HamzaShabbir517, 2022-08-30:
ok thanks i will rebuild in and let you know

HamzaShabbir517, 2022-08-30:
@TommyMurphyTM1234 Thanks its working completely fine now

TommyMurphyTM1234, 2022-08-30:
> @TommyMurphyTM1234 Thanks its working completely fine now

I presume you meant @zhongjuzhe  ? 🙂

TommyMurphyTM1234, 2022-09-14:
> @HamzaShabbir517 Yes, I'm glad to share my method to configure the toolchain.
>
> ```
> git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-next
> ```
>
> then, in .../riscv-gnu-toolchain/, switch [...]

Hi @HamzaShabbir517 - should these instructions still work? When I try them I get this:

```
user@hornbill:~/Downloads/rvv-toolchain$ git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-next
Cloning into 'riscv-gnu-toolchain'...
remote: Enumerating objects: 8099, done.
remote: Counting objects: 100% (26/26), done.
remote: Compressing objects: 100% (23/23), done.
remote: Total 8099 (delta 5), reused 15 (delta 2), pack-reused 8073
Receiving objects: 100% (8099/8099), 5.08 MiB | 7.63 MiB/s, done.
Resolving deltas: 100% (4023/4023), done.
user@hornbill:~/Downloads/rvv-toolchain$ cd riscv-gnu-toolchain/
user@hornbill:~/Downloads/rvv-toolchain/riscv-gnu-toolchain$ cd riscv-gcc/
user@hornbill:~/Downloads/rvv-toolchain/riscv-gnu-toolchain/riscv-gcc$ git checkout -b riscv-gcc-rvv-next
Switched to a new branch 'riscv-gcc-rvv-next'
user@hornbill:~/Downloads/rvv-toolchain/riscv-gnu-toolchain/riscv-gcc$ git pull
There is no tracking information for the current branch.
Please specify which branch you want to merge with.
See git-pull(1) for details.

    git pull <remote> <branch>

If you wish to set tracking information for this branch you can do so with:

    git branch --set-upstream-to=origin/<branch> riscv-gcc-rvv-next

user@hornbill:~/Downloads/rvv-toolchain/riscv-gnu-toolchain/riscv-gcc$ 
```

HamzaShabbir517, 2022-09-15:
@TommyMurphyTM1234 Yes but i think you dont need to git pull as when you checkout it checkout the branch with all the updates

Stark-qan, 2022-10-10:
> I think you should try to use --with-arch=rv32gcv in configure.

If I want my toolchain to support both RVV and RVB, am I using the right configuration option?
`../configure --prefix=$RISCV --with-arch=rv64gcv_zba_zbb_zbc_zbs_zbkb_zbkc_zbkx --with-abi=lp64d`