riscvarchive/riscv-gcc #355: What are the criteria for the test to pass?

Stark-qan, 2022-09-29:
We run our own test suite and find  many errors and failures.
I don't know if many errors and failures are negligible, and if there are some test cases that we can choose not to test (such as our tool chain for testing RISCV-gcc).

pz9115, 2022-09-29:
There are many testsuite in [gcc/testsuite](https://github.com/riscv-collab/riscv-gcc/tree/riscv-gcc-10.2.0/gcc/testsuite), and you can use dejagnu to run the regression test and see if there have some errors or failures. For riscv target, most testcase is in the [gcc/testsuite/gcc.target/riscv](https://github.com/riscv-collab/riscv-gcc/tree/riscv-gcc-10.2.0/gcc/testsuite/gcc.target/riscv), it use the option aim to riscv target.

After build the toolchain, you can use `make report-gcc -j$(nproc)` to run the test script, it will generate a report table show if there have some fail cases in gcc part. Some known fail cases was recorded at [test/allowlistc/gcc](https://github.com/riscv-collab/riscv-gnu-toolchain/tree/master/test/allowlist/gcc) in riscv-gnu-toolchain.

Stark-qan, 2022-10-10:
One more question about testing: the definition of a test case indicator in riscv-gcc/testsuits looks like this:
[image]
Where can I see a detailed explanation of them?
Specifically about this "dg-skip-if", shown below:
[image]
I got the following error when testing this use case, I don't know what the error and passing criteria are.
[image]