riscvarchive/riscv-gcc #357: Where are Callee-Saved registers present during tail-calls?

SaiVK, 2022-10-08:
Hello Everyone
  I am trying to understand how tail-calls work in the presence of callee-saved registers.
Say function _baz()_ calls _foo()_, which in turn performs a tail-call to function _bar()_. My question is, which one of the below cases would occur before the tail-call, whether
1.  The callee-saved registers w.r.t _baz()_ context is retained in the stack frame of _foo()_, which is reused by _bar()_. While _bar()_ returns to _baz()_, it will restore the callee-saved registers. 
2.  The callee-saved registers w.r.t _baz()_ context is restored to registers, and the stack frame of _foo()_ is popped out before the tail call occurs.  Now, again if these callee-saved registers are clobbered by the bar(), they would be saved and restored to the registers before returning to _baz()_. 

Which of the above two cases would be implemented while performing a tail call?

Thanking you
Sai

TommyMurphyTM1234, 2022-10-11:
Can't you look at the disassembly for compiled code to see what happens?

SaiVK, 2022-10-14:
Thanks @TommyMurphyTM1234 
 I understood it.

closed 2022-10-14 by SaiVK