riscvarchive/riscv-gcc #358: How can I make my GCC toolchain support both RVV and RVB?

Stark-qan, 2022-10-10:
Am I using the right configuration option?
`../configure --prefix=$RISCV --with-arch=rv64gcv_zba_zbb_zbc_zbs_zbkb_zbkc_zbkx --with-abi=lp64d`

TommyMurphyTM1234, 2022-10-11:
Have you tried it?
You'll have to build from the `rvv-next` branch for vector support.
```
git clone https://github.com/riscv/riscv-gnu-toolchain rv64-linux-rvv-next -b rvv-next
```

TommyMurphyTM1234, 2022-10-11:
For what it's worth, this toolchain build worked successfully for me but I don't have to hand any test vector, bitmanip, crypto code to test...
```
git clone https://github.com/riscv/riscv-gnu-toolchain rv64gcvb-newlib -b rvv-next
cd rv64gcvb-newlib/
./configure --prefix=`pwd`/installed-tools --with-arch=rv64gcv_zba_zbb_zbc_zbs_zbkb_zbkc_zbkx
date > build.log; make 2>&1 | tee -a build.log; date >> build.log
```

TommyMurphyTM1234, 2022-10-15:
@Stark-qan - can this issue be closed now? If so can you do it as I don't have privileges to do so in this repo? Thanks.

Stark-qan, 2022-10-19:
Yes, thank you for your help

closed 2022-10-19 by Stark-qan