riscvarchive/riscv-gcc #362: How can i add a custom insn in riscv.md?

Holypol, 2022-11-11:
I try to add a custom insn named "adds", which use to describe "a = b + c  - a".
in riscv.md, I write

`(define_insn_and_rewrite "adds<mode>4_internal"
  [(parallel
    [(set (match_operand:GPR          0 "register_operand" "=r,r")
          (plus:GPR (match_operand:GPR 1 "register_operand" " %r,r")
                   (match_operand:GPR 2 "add_operand"      " r,I")))
     (set (match_operand:GPR          3 "register_operand" "=r,r")
          (minus:GPR (match_dup 0)
                    (match_dup 3)))
    (clobber (match_scratch:GPR 4 "=X,X"))])]
  ""
  "#"
  "&& 1"
{
  emit_insn (gen_addssi3 (operands[0], operands[1], operands[2]));
  DONE;
}
  [(set_attr "type" "arith")
   (set_attr "mode" "<MODE>")])`

I just start to learn gcc, so  i try to add a new insn first to let gcc translate C code in md witch this insn .I have added this in binutils and gcc can identify successfully。

here is my simple test:
`int adds_sz1(int a, int b, int c){ a = b + c - a; return a;}
int adds_sz2(int a, int b, int c){ a = b + c; return a;}`

I know the gimple says that
`  _1 = b + c;
  a = _1 - a;
  D.1599 = a;`
I don`t know why my insn doesn`t work, could someone help me

kito-cheng, 2022-11-11:
You didn't post the pattern for `addssi3`, that would be the key to teach GCC how to synthesis instructions, in another word: you don't really need to write a `define_insn_and_rewrite` pattern, just get addssi3 with right RTL pattern, and then GCC will synthesis if possible...but it's hard to describe in a short comment.

But I would suggest you could send some time to read those slide first before continue your gcc hack: http://www.cse.iitb.ac.in/grc/gcc-workshop-12/index.php?page=slides , especially for the day 3 part, that is very helpful to understand how it work and how to write a md pattern in right way.

zhongjuzhe, 2022-11-11:
It seems that you want to use GCC "Canonicalization of Instructions".
Well, to use "combine" PASS combine the operation.

First, use -fdump-rtl-combine-details to check whether GCC has a chance to combine operations.
In your example, I saw this following information:
Failed to match this instruction:

(set (reg/i:DI 10 a0)
    (sign_extend:DI (minus:SI (plus:SI (subreg:SI (reg:DI 143) 0)
                (subreg:SI (reg:DI 144) 0))
            (subreg:SI (reg:DI 142) 0))))

According to this information, you can define a machine description like this:

**(define_insn ""
	[(set (match_operand:SI 0 "register_operand"       "=r")
	    (plus:SI (minus:SI (match_operand:SI 1 "register_operand"       "r")
	            (match_operand:SI 2 "register_operand"       "r"))
	        (match_operand:SI 3 "register_operand"       "r")))]
 ""
 "adds\t%0,%1,%2,%3")**

You can directly copy this machine description and try. 
It works fine and generate adds instruction.

For more details about GCC combining operation, please read this:
https://gcc.gnu.org/onlinedocs/gccint/machine-descriptions/canonicalization-of-instructions.html