riscvarchive/riscv-gcc #363: relation between rvv-submission-v1 branch, riscv-gcc-rvv-next branch and GCC upstream

xukl, 2023-01-04:
Hi @zhongjuzhe , thanks for the great work on RVV support. Could you kindly share with us about the status of branch rvv-submission-v1?

I see that your riscv-gcc-rvv-next and rvv-submission-v1 branch are the only active branches of this repo (other [branches](https://github.com/riscv-collab/riscv-gcc/branches) receive no commits for more than 1 year, and #352 shows "This repository is not used for active development anymore"). It seems that riscv-gcc-rvv-next branch is your workspace for RVV support in GCC according to #338, and you are commiting RVV to GCC upstream which we can find in [gcc-patches mailing list](https://gcc.gnu.org/pipermail/gcc-patches/). However I cannot find much information about your active-maintaining rvv-submission-v1 branch, thus I am confused.

* If I understand correctly, the name of "rvv-submission-v1" says it has something to do with upstreaming process, but what is the relation between the 3 GCCs exactly?
* And if I need to try out RVV in GCC, of the 3 GCCs, which should I use now?

zhongjuzhe, 2023-01-04:
I would suggest you use rvv-next which fully supported all features of RVV. rvv-submission-v1 is the branch that I am re-developing RVV base on the experience of rvv-next. rvv-submission-v1 is just a review platform for GCC maintainer @kito-cheng  to review and comment. Don't use it.  rvv-next should work fine basically in most of the cases, I think it's sufficient for you to use.

I can share my current work roadmap:

GCC13 support all latest RVV intrinsics.
GCC14 support auto-vectorization using RVV features.

The upstream RVV GCC is refined version base on the experience that I develop rvv-next. The performance of upstream RVV GCC is defintely much better than rvv-next and RVV LLVM.  For example, I rework the VSETVL PASS with a perfect algorithm (Lazy code motion) which can always generate optimzal vsetvli instructions.

xukl, 2023-01-09:
Wow, thank you! This perfectly solves my problem!

closed 2023-01-09 by xukl