riscvarchive/riscv-gcc #368: Unable to build TFLite with riscv toolchain

klchiu, 2023-02-28:
Hi,

I am trying to compile a Tensorflow lite code by using the riscv toolchain. My target is RISC-V [CVA6](https://github.com/openhwgroup/cva6) core. I use the [minimal example](https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/examples/minimal) as the starting point.

Here are the steps:

```
mkdir minimal_build
cd minimal_build
RISCVCC_PREFIX=$HOME/toolchains/riscv/bin/riscv64-unknown-linux-gnu-
cmake -DCMAKE_C_COMPILER=${RISCVCC_PREFIX}gcc -DCMAKE_CXX_COMPILER=${RISCVCC_PREFIX}g++ -DTFLITE_ENABLE_XNNPACK=OFF ../tensorflow/lite/
cmake --build . -j

```

However, it gives me errors like below.

```
riscv64-unknown-linux-gnu-g++: error: unrecognized command line option '-mavx2'
riscv64-unknown-linux-gnu-g++: error: unrecognized command line option '-mavx512f'
riscv64-unknown-linux-gnu-g++: error: unrecognized command line option '-mfma'
_deps/ruy-build/ruy/CMakeFiles/ruy_have_built_path_for_avx2_fma.dir/build.make:75: recipe for target '_deps/ruy-build/ruy/CMakeFiles/ruy_have_built_path_for_avx2_fma.dir/have_built_path_for_avx2_fma.cc.o' failed
make[2]: *** [_deps/ruy-build/ruy/CMakeFiles/ruy_have_built_path_for_avx2_fma.dir/have_built_path_for_avx2_fma.cc.o] Error 1
riscv64-unknown-linux-gnu-g++: error: unrecognized command line option '-mavx512vl'
CMakeFiles/Makefile2:5423: recipe for target '_deps/ruy-build/ruy/CMakeFiles/ruy_have_built_path_for_avx2_fma.dir/all' failed
make[1]: *** [_deps/ruy-build/ruy/CMakeFiles/ruy_have_built_path_for_avx2_fma.dir/all] Error 2
riscv64-unknown-linux-gnu-g++: error: unrecognized command line option '-mavx512cd'
riscv64-unknown-linux-gnu-g++: error: unrecognized command line option '-mavx512bw'
riscv64-unknown-linux-gnu-g++: error: unrecognized command line option '-mavx512dq'

```

I see these errors are related to the vector extension. The CVA6 core only supports I, M, A, and C extension. Could you let me know how to get around this issue or how to disable the V extension?

Thank you

linsinan1995, 2023-03-01:
These are options from X86, your host machine. This link https://www.tensorflow.org/lite/guide/build_cmake_arm have more info about how to cross-compile tf-lite.

klchiu, 2023-03-03:
I was able to compile it by adding some flags to the cmake like below.

```
RISCVCC_PREFIX=$HOME/toolchains/riscv/bin/riscv64-unknown-linux-gnu-
RISCV_CC_FLAGS="-funsafe-math-optimizations"

cmake -DCMAKE_C_COMPILER=${RISCVCC_PREFIX}gcc \
  -DCMAKE_CXX_COMPILER=${RISCVCC_PREFIX}g++ \
  -DCMAKE_C_FLAGS="${RISCV_CC_FLAGS}" \
  -DCMAKE_CXX_FLAGS="${RISCV_CC_FLAGS}" \
  -DTFLITE_ENABLE_XNNPACK=OFF \
  -DCMAKE_SYSTEM_NAME=Linux \
  -DCMAKE_SYSTEM_PROCESSOR=rv64 \
  ../tensorflow/lite
```

klchiu, 2023-03-03:
Problem solved

closed 2023-03-05 by klchiu