riscvarchive/riscv-gcc #374: error while compiling linux: compiler 'riscv64-unknown-linux-gnu-gcc' not found

appujee, 2023-04-03:
git branch
* (HEAD detached at v5.4)

Using instructions as described in: https://risc-v-getting-started-guide.readthedocs.io/en/latest/linux-qemu.html

I installed gcc on my ubuntu machine:
```sh
sudo apt install gcc-riscv64-unknown-elf`
```

System version
```sh
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux rodete
Release:	rodete
Codename:	rodete
```

```
+ make ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu- defconfig
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTLD  scripts/kconfig/conf
*** Default configuration is based on 'defconfig'
scripts/Kconfig.include:35: compiler 'riscv64-unknown-linux-gnu-gcc' not found
make[1]: *** [scripts/kconfig/Makefile:83: defconfig] Error 1
make: *** [Makefile:567: defconfig] Error 2

```

appujee, 2023-04-13:
> The Ubuntu package you installed includes the compiler binary `riscv64-unknown-elf-gcc`, not `riscv64-unknown-linux-gnu-gcc`.
> 
> If you want to use a Linux build, then the package is `gcc-riscv64-linux-gnu` (https://packages.ubuntu.com/search?keywords=gcc-riscv64-linux-gnu) and the compiler is invoked likewise.

thanks. i've added this to doc: https://github.com/riscv-admin/risc-v-getting-started-guide/pull/51 in case others get the same question.

TommyMurphyTM1234, 2023-06-13:
Can be closed now?

appujee, 2023-06-13:
Okay to close. btw,  can anyone approve the https://github.com/riscv-admin/risc-v-getting-started-guide/pull/51 ?

TommyMurphyTM1234, 2023-08-19:
> Okay to close. 

Could you close it so?
I think that only the issue originator or those with elevated privileges in this repo can close it.