riscvarchive/riscv-gcc #377: [issue][Stage 2][RISC_V] Build time issues during native build

akhileshkumar80, 2023-05-26:
Hi Experts 

I am trying to build native toolchain for my RISC-V target 
I am using https://github.com/riscv-collab/riscv-gnu-toolchain url  for  reference.  

Steps performed 
1-> git clone https://github.com/riscv/riscv-gnu-toolchain
2-> make linux 

Getting below link time issue during gcc stage2 . It seems some change are requited for native build because for cross build this repo is working fine.

please suggest how i can fix or build native toolchain for rics-v 

/home/user/opt/bin/ld: ./.libs/libisl.a(isl_aff.o): relocation R_RISCV_HI20 against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/home/user/opt/bin/ld: ./.libs/libisl.a(isl_aff_map.o): relocation R_RISCV_HI20 against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/home/user/opt/bin/ld: ./.libs/libisl.a(isl_affine_hull.o): relocation R_RISCV_HI20 against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/home/user/opt/bin/ld: ./.libs/libisl.a(isl_ast.o): relocation R_RISCV_HI20 against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/home/user/opt/bin/ld: ./.libs/libisl.a(isl_ast_build.o): relocation R_RISCV_HI20 against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/home/user/opt/bin/ld: ./.libs/libisl.a(isl_ast_build_expr.o): relocation R_RISCV_HI20 against `isl_aff_le_basic_set' can n

-------------------------------------------------------------------------
/home/user/opt/bin/ld: /home/user/riscv-gnu-toolchain/build-gcc-linux-stage2/./gmp/.libs/libgmp.a(gcd_22.o): relocation R_RISCV_HI20 against `__gmpn_clz_tab' can not be used when making a shared object; recompile with -fPIC
/home/user/opt/bin/ld: /home/user/riscv-gnu-toolchain/build-gcc-linux-stage2/./gmp/.libs/libgmp.a(hgcd_step.o): relocation R_RISCV_HI20 against `__gmpn_clz_tab' can not be used when making a shared object; recompile with -fPIC
/home/user/opt/bin/ld: /home/user/riscv-gnu-toolchain/build-gcc-linux-stage2/./gmp/.libs/libgmp.a(hgcd_appr.o): relocation R_RISCV_HI20 against `__gmpn_clz_tab' can not be used when making a shared object; recompile with -fPIC
/home/user/opt/bin/ld: /home/user/riscv-gnu-toolchain/build-gcc-linux-stage2/./gmp/.libs/libgmp.a(binvert.o): relocation R_RISCV_HI20 against `__gmp_binvert_limb_table' can not be used when making a shared object; recompile with -fPIC
/home/user/opt/bin/ld: /home/user/riscv-gnu-toolchain/build-gcc-linux-stage2/./gmp/.libs/libgmp.a(bdiv_q.o): relocation R_RISCV_HI20 against `__gmp_binvert_limb_table' can not be used when making a shared object; recompile with -fPIC
/home/user/opt/bin/ld: unresolvable R_RISCV_HI20 relocation against symbol `stderr@@GLIBC_2.27'
/home/user/opt/bin/ld: unresolvable R_RISCV_CALL relocation against symbol `strdup@@GLIBC_2.27'
collect2: fatal error: ld terminated with signal 11 [Segmentation fault]
compilation terminated.
make[5]: *** [Makefile:1696: isl_test_cpp] Error 1
make[4]: *** [Makefile:1997: all-recursive] Error 1

TommyMurphyTM1234, 2023-06-13:
> Steps performed 1-> git clone https://github.com/riscv/riscv-gnu-toolchain 2-> make linux
> 
> Getting below link time issue during gcc stage2 . It seems some change are requited for native build because for cross build this repo is working fine.

I don't understand - the steps outlined are missing any `configure` step
Apart from that, the steps seem to be for a cross build (e.g. building a toolchain that runs on x86 Linux that targets RISC-V).
So I'm not sure what you mean by "native build" here?
Besides, I have never experienced the errors mentioned in any of the many many builds of the tools that I have done.
You probably need to clarify the details of what you are doing to arrive at this situation.
Also - what OS are you using?

TommyMurphyTM1234, 2023-08-19:
Any update @akhileshkumar80?

akhileshkumar80, 2023-08-20:
Dear TommyMurphy 
sorry for delay response. 

for cross build no issue, for native build where host and target both are RTISC-V getting build time issues
but with some work around changes native build issues are resolved 

you can close this ticket now

TommyMurphyTM1234, 2023-08-20:
> you can close this ticket now

I cannot close it as I do not have rights.
Please close it yourself.
Thanks.

closed 2023-08-23 by akhileshkumar80