riscvarchive/riscv-gcc #379: about compressed instruction && RVV support

GMYMingyu, 2023-11-01:
Hi, all,
I currently working on the riscv-dv assembly generation to do the compressed and RVV instruction verification.
The rtl code is built on a modified 9.3.0 GCC. So I need the gloden GCC of 9.3.0.
I want to verif RVV instruction, but GCC 9.3.0[https://ftp.gnu.org/gnu/gcc/gcc-9.3.0](https://ftp.gnu.org/gnu/gcc/gcc-9.3.0/gcc-9.3.0.tar.gz) does not support RVV yet. So I upgrade the golden GCC to 11.1.0. However, GCC 9.3.0 and GCC 11.1.0 have different optimizations to compressed instructions. In other words, GCC9.3.0 may not compress one instruction but GCC 11.1.0 will compress it.

I would really appreciate it if anyone can give me some hints about the following questions:
1) Does there exist compressed optimization option for GCC? is it possible to set the instruction compression level?
2) Does GCC9.3.0 support RVV?

How I compile GCC9.3.0:
[image]

zhongjuzhe, 2023-11-02:
You mean you want to disable compressed instruction ? you can specify --with-arch = rv64gv

GMYMingyu, 2023-11-13:
> You mean you want to disable compressed instruction ? you can specify --with-arch = rv64gv

Thanks for your reply. 
I would like to turn off the optimization of the compressed instructions. So that only compressed instructions will be compressed.  In my tests, using more recent GCC versions resulted in the compression of more instructions(for a same assembly file). 
But I thought it is unrealistic. Anyway, I will close this issue.